motioneye-project/motioneyeos #506: C920 HD out of focus

KL501, 2016-09-13:
I installed motioneyeos  ( version 2016828 ) on my RP Gen1 ver:1.2.
Download of the iso file & copying all on the SD card was straight forward and worked straight away. First boot up it recognized the C920 and I was able to see the camera stream on one of my computers via LAN. But the picture is blurry , I did set the "Dimmers" to 100% but still no luck. I connected a second USB webcam (640x480) to the RP and with the mechanical focus I was able to get a nice focused picture ( video stream ). The C920 simply does not focus, the auto focus is not working. Can the auto focus function be turned On or OFF ? I would be even happy the set the focus manually somehow with some commands from the terminal direct on the RP ?   Many Thanks, KL501.

melashkov, 2016-10-02:
ssh to your pi,
```ssh
ssh admin@192.168.1.xx
```
where 192.168.1.xx is ip of you motion eye
run folowing:
```bash
uvcdynctrl -d video0 --set='Focus, Auto' 0
uvcdynctrl -d video0 --set='Focus (absolute)' 10
```
have a play with 'Focus (absolute)' to get perfect focus

if it works:
create file:/data/etc/userinit.sh
```bash
uvcdynctrl -d video0 --set='Focus, Auto' 0
uvcdynctrl -d video0 --set='Focus (absolute)' xxx
```

KL501, 2016-10-02:
Ah, very good, that worked well. I can now nicely adjust the focus. But the best focus is achieved with a value of 10 or bellow and this is at a object distance of only halve a meter from the camera. Objects that are further away will not focus very crisp ( I tried all values from 0 to 300). The biggest distance where the C920 can focus satisfactory is 1 meter or bellow. I think what I have to do is to open the camera and adjust the lens mechanically and I have seen other people doing it on you tube that had the same focusing problem. 
Thanks so much  melashkov , your advice was of great help !  KL501

melashkov, 2016-10-02:
Happy to help KL501 😄

closed 2017-06-01 by bortek

gfondeur, 2020-01-17:
how can I disable auto-focus?
I try the instructions from melashkov but it didn't work, probably because I have a newer release

IMBACalimba, 2023-12-09:
This will fix the problem in newer versions of montioneye

```
 v4l2-ctl --set-ctrl=focus_auto=0
 v4l2-ctl --set-ctrl=focus_absolute=30
```

First line will disable autofocus.
Second line configures manually the focus.