motioneye-project/motioneyeos #507: MotioneyeOS constantly crashing/dropping wifi with Element14 Wipi USB dongle

cleancodernz, 2016-09-14:
I am running the latest MotionEyeOS on the following setup

1. Raspberry Pi Model B (512Mb model with two USB ports)
2. Official Raspberry Pi foundation power supply (5V, 2A)
3. Official Raspberry Pi Camera V2
4. Element14 Wipi USB dongle (RALink 2870) http://nz.element14.com/element14/wipi/dongle-wifi-usb-for-raspberry/dp/2133900 

Everything gets detected on boot, but within about a minute the USB dongle loses connectivity. If I use a 1A power supply then the whole system is very unstable, leading me to believe it is a power or OS issue. Sometimes the Dongle is not even detected on boot.

Plugging in via ethernet does not have the same problem.

ccrisan, 2016-09-14:
How about using a 2A PSU?

cleancodernz, 2016-09-14:
The wifi dongle drops / disconnects / loses power with the official raspberry Pi 2A psu
With a 1A psu the system is even more unstable and I can't even use Ethernet

1A (Apple iPhone psu) = high instability and neither wifi or Ethernet stays stable

2A (raspberry Pi foundation psu) = only stable on Ethernet, wifi dongle only works for a short time.

Nothing is over clocked. I am using a static ip setup.

ccrisan, 2016-09-14:
Alright. Let's see a dmesg.log file after using the 2A PSU. The 1A PSU is definitely problematic and is out of question here.

cleancodernz, 2016-09-14:
Here are all the files from the system.

The fourth to last boot was actually successful with the wifi adapter. Then I tried to change the admin and user password.

The third and second to last boot the wifi adapter didn't seem to work

The last boot is on ethernet.

[MotioneyeOS.zip](https://github.com/ccrisan/motioneyeos/files/473354/MotioneyeOS.zip)

ccrisan, 2016-09-14:
From your log files I deduce the following two separate problems:

1. The OS has many times unexpectedly rebooted which is consistent with a PSU problem.

2. The WiFi connection goes down every now and then, judging from the following lines:

        Sep 14 10:56:41 meye-da5b2763 user.notice wifi: disconnected
        Sep 14 10:56:41 meye-da5b2763 user.notice network: wlan0 has no IP address
        Sep 14 10:56:46 meye-da5b2763 user.notice wifi: disconnected
        Sep 14 10:56:46 meye-da5b2763 user.notice network: wlan0 has no IP address
        Sep 14 10:56:51 meye-da5b2763 user.notice network: wlan0 has no IP address
        Sep 14 10:56:56 meye-da5b2763 user.notice network: wlan0 has no IP address
        Sep 14 10:57:01 meye-da5b2763 user.notice network: wlan0 had no IP address for 20 seconds, rebooting

    For the latter I'd try a different WiFi adapter (model) and see if it works. Also the 2 amps PSU might not be working as expected - try changing it as well. Another idea is to check to see if the DHCP server leases don't expire too soon (judging from the loss of IP address).

cleancodernz, 2016-09-14:
Cheers.

The rebooting happened much more when using the 1A PSU. On Ethernet it is very stable. It was also stable for a while this morning on wifi until I rebooted to apply some settings. Then it couldn't find the wifi adapter again. Is it possible to make it take longer to find the wifi adapter or is there some other reason why the wifi adapter is sometimes found on boot and other times now found?

ArmandH, 2016-09-22:
Just FYI i have exactly the same problem on Pi Zero, with motioneye the wifi disconnects at least once a day, when i only use streameye i don't have this problem (uptime 39d now) with exactly the same hardware.
I have not had time to investigate the cause, could be related that motioneye has higher load or is asking more of the wifi, no idea right now but as streameye is working perfect it doesn't seem related to hardware. I use official pi wifi dongle v2 camera and good powersupply.
I even created a script to reboot my pi zero when i can't login anymore from other pi, login sets counter due to the wifi problem with motioneye.

bortek, 2017-06-01:
these are HW issues not a software. Please google for raspberry pi stability and how to make it better.

closed 2017-06-01 by bortek

mtk64, 2018-06-29:
Hi, don't think that is a HW issue. My Raspian OS is running on a Zero W without any problems since months. I guess that might be a problem on WLAN auto power off: https://www.raspberrypi.org/forums/viewtopic.php?t=87801
Is there any config file to try this solution on motioneyeos?
kind regards
michael

maik-dmxc, 2018-09-16:
Hi,
i have the same problems with rpi zero w.
this rpi zero only make's trouble with motioneyeos, with rasbian all works fine.

regards maik