motioneye-project/motioneyeos #508: Install without wired connection

anow2, 2016-09-14:
I have no network cable lying around, and wanted to install MotionEyeOS.

I installed via Linux and set the wireless SSID and ppk, thinking this would be enough and a wired connection wasn't needed. 

When I boot up I get "starting wpa_supplicant failed (brcmfmac)" then an endless reboot.
I've read a few other threads on this, but had no luck. My wifi is set to WPA2.

Thanks!

ccrisan, 2016-09-14:
Well the only thing that comes to my mind is that your SSID/PSK are not entered correctly in your wpa_supplicant file. What does your wpa_supplicant.conf file look like? Do you have any special characters in your SSID/PSK?

anow2, 2016-09-14:
No special characters: this is the wpa_supplicant file:

update_config=1
ctrl_interface=/var/run/wpa_supplicant
network={
    scan_ssid=1
    ssid="YMH28"
    psk="wednesday12"
-e }

ccrisan, 2016-09-14:
Well everything looks fine except for the `-e` right before the ending bracket. Isn't that a typo?

anow2, 2016-09-14:
No, it wasn't, I could have sworn it was auto generated, let me remove and try it again

anow2, 2016-09-14:
Removed, and still not connecting. =/

ccrisan, 2016-09-14:
Where did you remove it from? I have a feeling you removed it from `/boot/wpa_supplicant.conf` whereas that file is only used as a template to generate the `/data/etc/wpa_supplicant.conf`.

anow2, 2016-09-14:
I don't have any of those directories, actually. I'm gonna go ahead and guess that's the problem? I've reimaged the SD card multiple times with different downloads of the same ISO, with the same structure.
[image]

ccrisan, 2016-09-14:
Auch you're on Windows. Well in that case, have you reimaged the SD card and used the fixed wpa_supplicant.conf?

anow2, 2016-09-14:
I just tried 20160705 instead of 20160828 and it ended up connecting.

And yes, I reimaged the other version and used the fixed wpa_supplicant. I have done this both on OSX (using the shell file) and Windows

ccrisan, 2016-09-14:
So what you're saying is that wifi support for the PI3 is broken in 20160828?

anow2, 2016-09-14:
Looks like it

ccrisan, 2016-09-14:
Well if it's not too much to ask, could you please try to upgrade to 20160828 and see if the problem is indeed a driver/hardware one and not a configuration-related one?

Qrishna, 2016-09-15:
I am trying to run the script writeimage.sh with my network ssid and key and I am getting endless reboots as well. Is there a way we can bypass these endless reboots without the ethernet cable?

Blue-Dust, 2016-09-25:
I had this same problem on the newest version.  After I set it up and got the boot loop because it wasn't able to find a NIC on my PiZero, I booted up my Pie2 on Raspbian and connected the MotionEyeOS to it via USB/Micro SD adapter.  I updated the wpa_supplicant.conf file, and it still didn't connect.  I installed 20160705, did the same thing, and it connected to my WiFi.  No issues.  I then upgraded it to the latest/greatest via web interface and it still connects/works as expected.

ccrisan, 2016-09-25:
@Blue-Dust ,

1. When you say NIC, do you mean a USB wifi adapter? When your Zero wasn't able to detect it, what was the error message during boot? Can you attach your boot.log file here?
2. Between 20160705 and 20160919 there's 20160915 which is supposed to be stable. Have you tried it? Does it work?

ItsJoeEnterprises, 2017-01-30:
I was able to get the Ethernet connection working but not the Wifi. I tried everything they offered in the forums, make a notepad file named wsp_supplicant.conf with network={ ssid="wifiname" psk="password"}. Nothing worked and just kept rebooting over and over again. I also could not move the file inside Rasbian (seperate SD card with Rasbian img installed). I opened up root terminal inside Raspbian and was able to make the conf file and I saved it to a USB drive and transferred it to Windows. When I wrote the IMG file onto the SD card on windows I moved the wsp_supplicant.conf file (That I made in root terminal in raspbian with network={ ssid="wifiname" psk="password"} inside) into the SD card before first boot (CCrisan stated that you need to do due it on the first boot and he was right). I was able to connect via Wifi first boot no issues. This worked for me and hopefully will work for you. 

It works on my rasberry pi 3 B but I can't get it to work on Pi Zero v1.3 so I will continue to try to make that work.

FaizB, 2017-02-16:
Hi before i give it a go. As a student in halls we have a portal page to log into so there is no protected network router password but instead you have to log in using a log in page. Does anybody know if there may be any issues. If not i can give it a go and report back.

bortek, 2017-06-01:
Closing this issue due to inactivity. If stil interested please post your question in the forum page.

closed 2017-06-01 by bortek