motioneye-project/motioneyeos #512: Not be able to reach the raspberry pi 2 after reboot

Eschtia, 2016-09-18:
Hello! 

After a reboot of my raspberry pi connected only via ethernet it isn't reacheble my router does not show my raspberry.

Raspberry Pi 2 with a Microsoft Webcam and connected via ethernet
The problem persists after a fresh install too.

I can't send a log file because it is not reacheble via web and via ssh

I hope someone can help me

---sorry for my bad english--- :(

ccrisan, 2016-09-19:
What version is this? Have you tried other OSes on the same Raspberry PI unit? Can you connect a monitor on the HDMI output and see what's going on?

Eschtia, 2016-09-19:
I tried many different builds. The problem was the SD card i changed to an other one and now its ok

thank you

ccrisan, 2016-09-19:
Alright I'll close the issue now.

closed 2016-09-19 by ccrisan