motioneye-project/motioneyeos #513: Higher video resolution supported with Raspberry Pi Camera Board V2

Shineson1001, 2016-09-18:
Video resolution 1920x1440 works fine with the Raspberry Pi Camera Board V2 (https://www.modmypi.com/raspberry-pi/camera/raspberry-pi-camera-board-v2-8mp1080p)
Would be nice, if you provide this video resolution in the web frontend.

ArmandH, 2016-09-22:
Also streameye you can't set higher then 1080

raspimjpeg.py: error: option height: must be between 64 and 1080

SkyByte, 2016-09-23:
It would at least be nice to be able to select the video resolution for the still images separately from the video streaming resolution, incl. the 3280x2464 resolution for the new Raspberry Pi Camera Board V2.

In my personal case, i would like to take some still images with the highest resolution as possible, but i don't need a high video stream resolution (even not simultaneously). How about a additional menu entry at 'Still Images' to select the 'Still Image Resolution'?

jbthoma, 2016-10-30:
Yes, I also have the V2 camera and agree -- it would be nice to take still images at the full 3280x2464 resolution.

ccrisan, 2016-11-16:
Fixed with https://github.com/ccrisan/motioneye/commit/f1298c012312aee69e231d2831abab2c3d76a0a6 and https://github.com/ccrisan/motioneye/commit/d61858b0ad47a13428e850ee475619ac57e0af62.

closed 2016-11-16 by ccrisan