motioneye-project/motioneyeos #514: Trying to get wifi configuration into Pi Zero image

Dazzathedrummer, 2016-09-19:
Hi, 
I've got MotionEyeOS working on a Pi 3 and it functions perfectly. I would like to use the Pi 3 as a hub and then have several Pi Zeros suppling feeds to the hub.

I'm having some trouble getting the wifi configuration set up for the Zero, obviously, as there isn't a ethernet port, I am trying to follow your method of using the writeimage.sh script with the following...

`./writeimage.sh -d /dev/mmcblk0 -i "/Users/darrenb/Downloads/motioneyeos-raspberrypi-20160915.img"  -n 'TescoBroadband123:MyActualPassword' 
`
...but I am getting '-bash: ./writeimage.sh: No such file or directory'
or './writeimage.sh: command not found' if I use sudo.

Both the image and the writeimage.sh are located in my downloads folder (MacBook Pro, El Capitan) - I'm not too hot with Terminal so perhaps I'm doing something wrong?

Dazzathedrummer, 2016-09-19:
...Ok, I managed to get it to run, but now I get the following....

`:: writing disk image to sdcard
dd: /dev/mmcblk0: Operation not supported
:: mounting sdcard
Unmount failed for /dev/mmcblk0
mount_msdos: /dev/mmcblk0s1: No such file or directory
:: creating wireless configuration
:: unmounting sdcard
umount: ./.boot: not currently mounted
rmdir: ./.boot: Directory not empty`

?

ccrisan, 2016-09-19:
I won't be able to help you if you're on OSX. At this point I recommend writing the OS image the regular way (using whatever tool there is on OSX to write OS images in general), and then manually creating a `wpa_supplicant.conf` file placed on the boot partition. See [this](https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting-Raspberry-PI-Models-A-%28Plus%29-And-Zero) for more details.

Dazzathedrummer, 2016-09-19:
Excellent - that works perfectly!

Dazzathedrummer, 2016-09-19:
....just for clarity, I'll just add that the wpa_supplicant.conf needs to be dropped into the SD card before the first boot.
I Initially tried this using an image that I had already installed - but of course, the wpa_supplicant.conf was ignored.

thanks for your help!

ccrisan, 2016-09-19:
I'm glad I could help.

closed 2016-09-19 by ccrisan