motioneye-project/motioneyeos #515: Not getting feed from remote camera

Dazzathedrummer, 2016-09-19:
HI again,

So, I now have a working Pi3 hub and a remote Pi Zero camera. For some reason, I'm not getting a consistent feed from the Zero - If I browse directly to the Zero, I get a perfect feed - but if I go to the hub, very often, the camera is not visible - I'm not sure why this is? is there anything in the settings that I need to tweak?

Pi Zero feed from 192.168.1.80
[image]

Pi3 Hub at 192.168.1.69
[image]

ccrisan, 2016-09-19:
Please attach the motioneye.log file.

Dazzathedrummer, 2016-09-19:
File attached...
[motioneye.log.zip](https://github.com/ccrisan/motioneyeos/files/480191/motioneye.log.zip)

ccrisan, 2016-09-19:
Well it seems there's a communication problem between the two:

    ERROR: failed to get current picture for remote camera 1 on http://192.168.1.80/config/1: request timed out

Most likely the network bandwidth between the two is not enough for the selected resolution and framerate. I recommend reducing the framerate and/or the resolution.

Dazzathedrummer, 2016-09-19:
Ok, thanks - I'll give that a go.

closed 2017-06-01 by bortek