motioneye-project/motioneyeos #520: Add network monitor command to motioneyeos - suggest include attached "delta" program

doughadfield, 2016-09-22:
Hi, I've found the need to monitor network traffic out of the motioneyeos platform, when testing various settings. The current image only has "top" and "ifconfig" to monitor the system; tools like dstat are missing (clearly the need to keep the image size down means keeping such tools to a minimum). However it would be really useful to have a network monitor tool of some sort in the image. If you can't fit "dstat" (I know that "dstat" is supported by busybox) then I found the attached python program "delta" on the web. I can't remember the author (and can'find where I downloaded it from). I've modified it slightly to report in "k" rather than bytes, but it means you can do things like "delta ifconfig" or "delta cat /sys/class/net/eth0/statistics/tx_bytes", which is precisely the value I was looking for.

Perhaps you could include this delta tool in /usr/bin in the standard image, in future releases (if you can't fit dstat in!!).

thanks again
Doug
[delta.txt](https://github.com/ccrisan/motioneyeos/files/486866/delta.txt)

ccrisan, 2016-09-22:
To be honest, I find having such a monitoring tool on motionEyeOS not very useful for most of the regular users. People that wish to monitor parts of their system can always write their own scripts and place them somewhere in `/data`.

The Raspberry PI version of the OS has UI options to enable system monitoring. A script is automatically created to this end. This is how it's generated: https://github.com/ccrisan/motioneyeos/blob/master/board/raspberrypi/motioneye-modules/boardctl.py#L30.

doughadfield, 2016-09-22:
Thanks for your response, Calin. I'll look into the script you mention - do I run the script interactively from the root ssh login, or does it run automatically and produce some sort of batched output?

ccrisan, 2016-09-22:
If you have a Raspberry PI, just enable monitoring in expert settings. Then you'll find the script at /data/etc/monitor_1 (IIRC).

doughadfield, 2016-09-22:
Excellent, thanks. I haven't experimented with the monitoring area yet, so that'll be interesting. Yes, I'm exclusively (at the moment) Pis. I have motioneyeos systems running on zero, ModA and Mod3 systems. Typically, I've found motion detection is only practical on the big Pis (mod2, mod3) so all my smaller Pis are using fast network camera mode (which runs really well even on old ModA 256MB systems). I have a C.H.I.P on order, so it'll be interesting to see what that does with Motioneyeos :-)

closed 2017-06-01 by bortek