motioneye-project/motioneyeos #521: I want to continuously record, but on a schedule. Can't seem to record any footage without Motion Detection turned on.

dev67, 2016-09-22:
When I turn off motion the schedule disappears, and nothing records even though I still have continuous selected in recording mode under the movie section.

ccrisan, 2016-09-22:
Please attach your config files (after removing sensitive data from them). You can download them as a backup archive from the UI, then extract the files called thread-*.conf.

dev67, 2016-09-26:
It seems like files are getting saved continuously like the settings say they should to my external 1 TB drive. But that clips are only getting uploaded to my google drive when motion occurs. I'm looking to have all continuous footage uploaded. Sometimes I'm getting a lot of dropped frames too. My movie files are set max size to 600 seconds (10 minutes) but sometimes the video clips are as low as 7:40. Looked at my time stamp in the bottom corner and it shows a full 10 minutes. Not sure if this is a codec thing or what? 

[motioneye-config.zip](https://github.com/ccrisan/motioneyeos/files/493540/motioneye-config.zip)

dev67, 2016-10-03:
You can close / skip this issue, it's resolved. Thank you though!

closed 2016-10-03 by ccrisan