motioneye-project/motioneyeos #523: 20160919 - Update failed error

stevec001, 2016-09-24:
When trying to upgrade from version 20160915 to latest Pre-release ver 20160919 I receive an 'update failed' error and have to reset power because the system locks up.  This is with a PI 3 and pi cam v2.  Has anyone else had similar results?

stevec001, 2016-09-24:
Here is the MotionEye log file clip from the upgrade attempt. 
[MotionEyeLog.txt](https://github.com/ccrisan/motioneyeos/files/491176/MotionEyeLog.txt)

ccrisan, 2016-09-24:
Did you reset the device while downloading the update? At `20:37:50` it says

    2016-09-21 20:37:50: [motioneye]     INFO: downloading https://github.com/ccrisan/motioneyeos/releases/download/20160919/motioneyeos-raspberrypi3-20160919.img.gz...

and then the next line is from 5 minutes later, saying

    2016-09-21 20:42:31: [motioneye]     INFO: hello! this is motionEye server 20160915

This usually happens if you reset the device while updating.

stevec001, 2016-09-24:
I did not physically did not do anything till after receiving the failed message. It seams to not want to update now. Did the SD card become corrupt?

ccrisan, 2016-09-24:
It is possible. However I tend to incline towards an unwanted reset. Please attach the rest of your logs to see if we can find out the reason of the reset.

stevec001, 2016-09-24:
Will do when I have local access to the device later today.

stevec001, 2016-09-24:
Here is the log file for Sept to date. Should show the previous update.  I have tried this pre-release update a few times all with the fail error.  Any insight would be great.
[motioneye_log_sept_to_date.txt](https://github.com/ccrisan/motioneyeos/files/491404/motioneye_log_sept_to_date.txt)

stevec001, 2016-09-25:
I think I closed by accident when posting update.

ccrisan, 2016-09-25:
Could you please attach *all of your logs*?

stevec001, 2016-09-25:
Here you go.  

[messages.log.txt](https://github.com/ccrisan/motioneyeos/files/491678/messages.log.txt)
[motion.log.txt](https://github.com/ccrisan/motioneyeos/files/491679/motion.log.txt)
[boot.log.txt](https://github.com/ccrisan/motioneyeos/files/491680/boot.log.txt)
[motioneye.log.txt](https://github.com/ccrisan/motioneyeos/files/491681/motioneye.log.txt)
[dmesg.log.txt](https://github.com/ccrisan/motioneyeos/files/491682/dmesg.log.txt)

ccrisan, 2016-09-26:
All of your logs are consistent with a power loss during update. I'm sorry but I have no other ideas regarding your issue.

stevec001, 2016-09-27:
Calin, I tried the updated once again, I received the "update failed" message again.  Instead of rebooting the device right away when it seemed unresponsive, I waited.  It took about 15 minutes and I was able to log on finally. The update was successful even though I received the failed message. We will see what happens in the future.  Thank you for your help in this mater. I never had an update act in this manner before. Great program. Please keep up the great work!

ccrisan, 2016-09-27:
My guess is that the download was too slow. It is usually done within 1 minute, but given that the image file is about 65M, on slower connections it may pose a problem.

closed 2017-06-01 by bortek