motioneye-project/motioneyeos #528: How to geht the Stream (rtp/rtsp) on HDMI output of my Pi3-ModB?

ted69mosby, 2016-09-26:


doughadfield, 2016-09-26:
Hi, there's a simple way to do this, and it's being included in the motioneyeos in the fullness of time. In the meantime, check out the latest version of the raspimjpeg.py script in the "https://github.com/ccrisan/streameye" git repository (under the "extras" subdirectory). This script has an additional argument "--preview" which will enable the camera output to the HDMI display port. Add this extra argument to the raspimjpeg.py execute line in the /usr/bin/streameye.sh script (the root file system will need to be re-mounted read/write, so the script can be modified). You could add the "preview" argument to the /data/etc/raspimjpeg.conf file, but that will be overwritten when if the GUI is used subsequently to update the video camera settings.

enjoy
Doug

bortek, 2017-06-01:
Closing his issue due to inactivity. If stil interested please post your question in the forum page.

closed 2017-06-01 by bortek

Tobser44, 2018-11-27:
> Hi, there's a simple way to do this, and it's being included in the motioneyeos in the fullness of time. In the meantime, check out the latest version of the raspimjpeg.py script in the [...]

Hello Doug,

I just found your post and I try to do what you mentioned. I'm new into programming and try to understand your words.

I looked up for the "--preview" mark in raspimjpeg.py and found this:

```
parser.add_argument('-p', '--preview', help='enable camera preview to HDMI port', 
        action='store_true', dest='preview', default=False) 
parser.add_argument('--preview-width', help='preview width, in pixels (64 to 3280, if different from capture width)', 
        type=int, dest='preview_width', default=False) 
parser.add_argument('--preview-height', help='preview height, in pixels (64 to 2464, if different from capture height)', 
        type=int, dest='preview_height', default=False) 
```

So, if I got you wright, I need to add the "-p" argument to streameye.sh. So I tried to understand the code, but I'm not sure, where to add. 

Can you please help me with that?

Just to get the preview function wright, it allows me to see the video via HMDI on a TV, but I'm still able to save videos and pictures? 

I found another post of you, where you described how to activate HDMI preview in fast network mode. But that's not what I want.

doughadfield, 2018-11-27:
Hi, I think you should be able to do what you want by switching on the “preview” switch in the camera GUI. Calin has integrated the preview mode fully into the motioneyeos GUI interface now, so you shouldn’t need to modify any scripts. 

 

Try it and let us know how you get on.

 

                Doug

Tobser44, 2018-11-28:
Hey Doug, thanks for the reply.

The HDMI preview option is only available when you use "Fast Network Camera". I haven't found the option when "Fast Network Camera" is OFF. If I'm wrong, let me know :-)

Thanks Tobias