motioneye-project/motioneyeos #534: Dropbox upload question

andy6149, 2016-09-30:
Where do I need to drop files to get motion eye to upload to Dropbox? Reason I ask is that I am FTPing videos from ip cam, as I get better quality video, to the motion eye which already has a working Dropbox upload but I don't know where to drop the video thanks

ccrisan, 2016-10-01:
You are not supposed to "drop" files anywhere. motionEyeOS will upload movies and pictures automatically at the moment they are created.

andy6149, 2016-10-01:
Yep but I find the frame rate too slow so I have the camera ftp the movies to the motion eye folder the for the camera. The motion eye shows the movies in the right place and I can view them correctly but they don't upload to drop box, I had thought it was some kind of cron job that did the uploading

closed 2017-06-01 by bortek