motioneye-project/motioneyeos #535: RaspberryPi Zero Camera Not Detected

webunraveling, 2016-10-01:
Before I get started, I have to say, nice work on this project. I just setup my RaspberryPi Zero (1.3) with the lastest version of MotionEyeOS (motioneyeos-raspberrypi-20160919). After finding out that I needed a wifi connection with a password, I got connected and everything works great... except **there is no camera detected**. I have tried shutting down and reconnecting the camera but it didn't make a difference. It appears to be connected fine. Did I miss something for cameras on the RaspberryPi or should it work out of the box?

Here are the log files. In case something is going wrong other than me not being able to find an answer in searches.

* [motioneye log](http://pastebin.com/rFdAHyqa)
* [dmesg](http://pastebin.com/G9V4nfzf)
* [boot log](http://pastebin.com/MaB11Cjk)
* [messages](http://pastebin.com/TDN2kDjL)

ccrisan, 2016-10-01:
Does the camera work on your Zero with Raspbian?

webunraveling, 2016-10-11:
Alright, sorry for not responding right away. I had to order a powered USB hub to be able to plugin a keyboard or mouse while running Raspbian. It is not detected on Raspian. I have a few more Pi Zeros on the way, along with camera. Hopefully I can do some testing on other devices then. I've tried some troubleshooting I've found on forums but none have provided a solution yet.

closed 2017-06-01 by bortek