motioneye-project/motioneyeos #537: RPI 3 onboard Wifi not working

NewbPi, 2016-10-02:
Hey, Just found this github, and it seems to suit my needs.

But after setting up the settings with RJ45 cable plugged in. And then rebooting without the cable. The Wifi dosnt seem to come on.

Here is some info in hope to get help with this problem.

Version; motioneyeos-raspberrypi3-20160919
Wifi security: WPA2-AES
Password is Correct
SSID is correct.
Tried both DHCP and Manual setup of IP

ccrisan, 2016-10-02:
Please attach your boot.log here. You can download it from the UI, using an ethernet connection.

NewbPi, 2016-10-02:
---- booting motionEyeOS 20160919----
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting watchdog: done
 * Starting eudev: done
 * Setting root password: done
 * Starting wpa_supplicant: failed (brcmfmac)
 * Configuring wireless network: no link
 * Configuring wired network: no link

NewbPi, 2016-10-03:
Anyone?

ccrisan, 2016-10-03:
Your adapter is detected by the OS but the wireless connection cannot be established. It's difficult to tell what the exact problem is. Does it work with any previous stable releases?

NewbPi, 2016-10-04:
---- booting motionEyeOS 20160915----
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting watchdog: done
 * Starting eudev: done
 * Setting root password: done
 * Starting wpa_supplicant: failed (brcmfmac)
 * Configuring wireless network: no link
 * Configuring wired network: dhcp
 * Setting current date using http: Tue Oct 4 12:01:31 UTC 2016
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Starting motioneye: done
 # Interface eth0 has IP address 192.168.1.19/24
 # Default gateway is 192.168.1.1
 # DNS server address is 192.168.1.1

ccrisan, 2016-10-04:
Can you post your `/data/etc/wpa_supplicant.conf` here? You can change your password into something else before posting it but I'd like to know if you have any special characters in it.

NewbPi, 2016-10-04:
How do i access it?

ccrisan, 2016-10-04:
Using [ssh](https://en.wikipedia.org/wiki/Secure_Shell) or [putty](http://www.chiark.greenend.org.uk/~sgtatham/putty/download.html). See also https://github.com/ccrisan/motioneyeos/wiki/FAQ#how-can-i-log-in-on-sshtty-console.

NewbPi, 2016-10-04:
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
    scan_ssid=1
    key_mgmt=WPA-PSK WPA-EAP
    psk="penispump"
    ssid="pewpew"
}

NewbPi, 2016-10-04:
Can the problem be that my router is running WPA2-PSK [AES]

ccrisan, 2016-10-04:
How did you get that key_mgmt there? Was it automatically generated somehow?

NewbPi, 2016-10-04:
It was there when i went into the file? I havnt added it.

ccrisan, 2016-10-04:
Have you upgraded from a previous version or is this a brand new installation of 20160915/20160919?

NewbPi, 2016-10-04:
Brand new installation downloaded from this site

NewbPi, 2016-10-04:
Also tried version motioneyeos-raspberrypi3-20160705

same problem.

ccrisan, 2016-10-04:
Well have you removed the key_mgmt line from wpa_supplicant.conf to see if it works?

NewbPi, 2016-10-05:
Removed the "key_mgmt=WPA-PSK WPA-EAP" part but still no connection.

---- booting motionEyeOS 20160915----
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting watchdog: done
 * Starting eudev: done
 * Setting root password: done
 * Starting wpa_supplicant: failed (brcmfmac)
 * Configuring wireless network: no link
 * Configuring wired network: dhcp
 * Setting current date using http: Wed Oct 5 06:13:18 UTC 2016
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Starting motioneye: done

ccrisan, 2016-10-05:
Does it work with Raspbian and this exact wpa_supplicant.conf file?

NewbPi, 2016-10-06:
How do i create it on raspbian? It says that the file dosnt excists.

ccrisan, 2016-10-06:
https://www.raspberrypi.org/documentation/configuration/wireless/wireless-cli.md

NewbPi, 2016-10-07:
The wpa file from motioneyeios dosnt work on raspbian.

I have to make it look like this to get it to work.

country=DK
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
    ssid="ssid"
    psk="password"
}

premingiet, 2016-10-14:
unable to connect to inbuilt wifi of raspberry pi 3
on seeing the boot.log file, it said:

"Starting wpa_supplicant: failed (brcmfmac)
Configuring wireless network: no link"

my /data/etc/wpa_supplicant.conf file is

update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
scan_ssid=1
psk="TARGET"
ssid="9437945909"
}

But when i used raspberry pi jessie OS, i was able to use wifi very easily.
There my wpa_supplicant.conf file was

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
ssid="TARGET"
psk="9437945909"
}

i am using motioneyeos-raspberrypi3-20160919.img

whats the issue ?

navigonic, 2016-11-06:
Same problem here, works well with wired but not with wifi, wpa_supplicant looks like the first poster (except no KEY_MGT)

ccrisan, 2016-11-06:
@navigonic what version of motionEyeOS? Can you please attach your log files?

rbelew, 2016-12-09:
@CCrisan et al, i seem to be having this same issue, with motioneyeos-raspberrypi3-20161125.img.  

* RPI3 wireless hardware works fine with standard raspbian motioneyeos-raspberrypi3-20161125
* preparing the SD card specifying my WAP details as `-n` command arguments causes reboot loop
* preparing the SD card without this parameter and a direct ether cable allows MEye to boot and seems to work (eg, can see camera via :80 access)
* I modified the `/data/etc/wpa_supplicant.conf` file as described in [Setting up WiFi doc](https://www.raspberrypi.org/documentation/configuration/wireless/wireless-cli.md)
* since `ifdown` command isn't available, i reboot
* on attempt to reboot, i get a series of messages as it tries wifi ports`:00:00:16, :21,36,41,46,51,... :06  wifi: disconnected`

Any suggested next debugging steps?

ericbrandon, 2016-12-10:
I had the exact same problem.  On RP3, motioneye version 20161125 worked fine over ethernet, but was in an endless reboot loop with wifi.

I solved the problem by editing /data/etc/wpa_supplicant.conf

Originally, some combination of the motioneye installation script and using the settings interface while connected by ethernet had created the following /data/etc/wpa_supplicant.conf:

update_config=1
ctrl_interface=/var/run/wpa_supplicant
network={
    scan_ssid=1
    ssid="mynetworkssid"
    psk="mypassword"
-e }

I editing the file to read like this:

update_config=1
ctrl_interface=/var/run/wpa_supplicant
network={
    ssid="mynetworkssid"
    psk="mypassword"
}

Now it works fine.

Thanks for a great piece of software!!

rbelew, 2016-12-10:
thanks @ericbrandon and i tried your edit, but no love for me.

the error messages are "<13>*timeStampHere*: wifi disconnected"

i've tried this both with and without the ether cable plugged in (because i think i read @CCrisan saying somewhere that this only supports one interface?)

the tedious part of this debug loop is that once i try to enable the wpa_supplicant, i seem always stuck in the endless reboot loop, and need to go back a write a new SDcard every time(:

ericbrandon, 2016-12-10:
Unfortunately, @rbelew,  I never had the error message you did.  My error message was the same as the original poster's:
Starting wpa_supplicant: failed (brcmfmac)

rbelew, 2016-12-10:
yes, sorry, i was not complete: i also get OP's error message, followed by the others i mentioned.

ccrisan, 2016-12-10:
@ericbrandon can you reproduce the problem with the "-e" extra text? I would really like to fix that.

@rbelew,

> on attempt to reboot, i get a series of messages as it tries wifi ports:00:00:16, :21,36,41,46,51,... :06

This doesn't make any sense. What is a "wifi port"? Are those wifi 802.11a channels?

Could you please post your wpa_supplicant.conf file contents as they are right now?

rbelew, 2016-12-10:
@ccrisan , sorry for my first report.  my second report mentions these were 
        <13>timeStampHere: wifi disconnected
messages, which i didn't initially recognize because the clock hadn't been set and they all had a `00:00:` prefix

my  `wpa_supplicant.conf` file is identical to the one @ericbrandon said worked for him.

i'm now working from a raspbian OS and your instructions for installing [MotionEye on Raspian](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian)

ericbrandon, 2016-12-11:
I'm not really able to rebuild & reinstall Motioneye from scratch right now because I don't have an extra RPI3 sitting around (they're all in use, right now).

But when I get a new one I will try to reproduce the steps that created the extra "-e" in the wpa_supplicant.conf file.

In the meantime, I can let you know that all I did was:
1) Download the latest build (20161125)
2) Build it on a card using your script and the -n flag, with simple no-special-character SSID and Password fields.
3) WiFi wasn't working, so tried a bit to edit the Wifi parameters using the Motioneye UI while connected by ethernet.  I don't remember the details of what I tried here.

noscoot, 2017-01-08:
I am having the same issues.  I found that my wifi password was too long.  I was using a full length (64 chars) password, but when I shorten it 10 chars it all worked. 

This is on a pi3 running
motionEye Version	20161212
Motion Version	4.0.1+git37b3595

also thanks Calin for your awesome work.

noscoot, 2017-01-08:
after more investigation, I think it might be it doesn't like number within the password.  Seems strange, but it working with 63 password without integers.

Something to try.

douggranger, 2017-02-19:
I am having the same problem.   Password and SSID are correct.  If i install raspbian the wifi works fine. 

--- booting motionEyeOS 20170212----
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting watchdog: done
 * Starting eudev: done
 * Setting root password: done
 *** Starting wpa_supplicant: failed (brcmfmac)**
 * Configuring wireless network: no link
 * Configuring wired network: dhcp
 * Setting current date using http: Sun Feb 19 12:28:18 CST 2017
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Setting smb admin password: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 # Interface eth0 has IP address xxx.xxx.xxx.xxx/24
 # Default gateway is xxx.xxx.xxx.xxx
 # DNS server address is xxx.xxx.xxx.xxx

ccrisan, 2017-02-19:
Does this work with any of the previous releases of motionEyeOS?

michael24h, 2017-02-19:
Try this

$sudo nano /etc/network/interfaces

auto lo

iface lo inet loopback
iface eth0 inet dhcp

allow-hotplug wlan0
auto wlan0

iface wlan0 inet dhcp
        wpa-ssid "ssid"
        wpa-psk "password"

ccrisan, 2017-02-19:
@michael24h this is definitely not going to work on motionEyeOS.

douggranger, 2017-02-19:
@ccrisan I have not tried any other build.  i just downloaded the most recent.

ccrisan, 2017-02-19:
@douggranger please try a few stable releases and see if it works. It might be a regression bug.

douggranger, 2017-02-19:
i will see what i can do. thanks

eenlow, 2017-02-26:
I had this issue and found I was entering the SSID with all lower case and I matched it exactly and things are working on a pi 3

studioBeaver, 2017-04-19:
@ericbrandon 
Thanks for your solution, this also worked for me:
update_config=1
ctrl_interface=/var/run/wpa_supplicant
network={
ssid="mynetworkssid"
psk="mypassword"
}
I had multiple empty entries in /data/etc/wpa_supplicant.conf, after I cleaned that up, like you sugessted it connected without any probles. Before I was stuck in reboot loop if I dissconected the LAN cable.
Maybe I made this problem myself, because I frist enable wireless without any values in the fields(SSID / pass) and clicked apply.
I was using motioneyeos-raspberrypi3-20170329.img.gz release.

closed 2017-06-01 by bortek