motioneye-project/motioneyeos #539: Help with file storage and FTP

artsymike, 2016-10-02:
I have the raspberry pi B and my internal storage is full and now i can no longer view the stream and cant see nor remove photos and videos stored on the SD card. Can anyone help please. I'm always forced to reinstall the whole OS every time this happens.

Mike.

Zaptoid, 2016-10-03:
Hi

Have you tried to SSH into the Pi and remove/move the images & videos? 

If you're unsure as to how, I can post instructions if need be.

artsymike, 2016-10-09:
I'm very new to this so yes if it's not to much trouble can you show me how to SSH into the pi and delete the photos and videos I ha e stored on to it? Thank you very much

Mike

samiultahir, 2017-03-14:
I have the same problem!
WinSCP won't even let me access it because when I type the correct username and password, it just says access denied.
Can someone please help!

closed 2017-06-01 by bortek