motioneye-project/motioneyeos #544: No local Camera detectet

dontspammexyz, 2016-10-05:
I'm running motioneyeos-raspberrypi3-20160919.img on an Raspi 3 with standard CSI Camera. The red light of the camera is on.

> lsmod
> Module                  Size  Used by    Not tainted
> ipv6                  334070  0 [permanent]
> brcmfmac              175239  0
> brcmutil                5585  1 brcmfmac
> cfg80211              416114  1 brcmfmac
> rfkill                 14461  1 cfg80211
> bcm2835_gpiomem         2939  0
> uio_pdrv_genirq         3089  0
> uio                     7715  1 uio_pdrv_genirq
> bcm2835_wdt             3128  1
> bcm2835_v4l2           37186  0
> videobuf2_vmalloc       5054  1 bcm2835_v4l2
> videobuf2_memops        1249  1 videobuf2_vmalloc
> videobuf2_v4l2         15516  1 bcm2835_v4l2
> videobuf2_core         19772  2 bcm2835_v4l2,videobuf2_v4l2
> v4l2_common             3758  2 bcm2835_v4l2,videobuf2_v4l2
> videodev              130637  4 bcm2835_v4l2,videobuf2_v4l2,videobuf2_core,v4l2_common
> media                  11168  1 videodev

> ls /deb/video0
> ls: /deb/video0: No such file or directory

What is going wrong and how to fix it up?
[image]

ccrisan, 2016-10-05:
Are you sure the camera isn't already added?

dontspammexyz, 2016-10-05:
[image]
Also I did shutdown, disconnected the cable, and readded it and turn power on. The same issue: no camera.

ccrisan, 2016-10-05:
Does it work with raspbian?

dontspammexyz, 2016-10-05:
On 2016-09-23-raspbian-jessie
there is no /dev/video0 , too.

> root@raspberrypi:~# modprobe bcm2835_v4l2

> root@raspberrypi:~# ls /dev/|grep vid
> root@raspberrypi:~# lsmod
> Module                  Size  Used by
> bcm2835_v4l2           38812  0
> videobuf2_vmalloc       5187  1 bcm2835_v4l2
> videobuf2_memops        1268  1 videobuf2_vmalloc
> videobuf2_v4l2         16535  1 bcm2835_v4l2
> videobuf2_core         20861  2 bcm2835_v4l2,videobuf2_v4l2
> v4l2_common             3929  2 bcm2835_v4l2,videobuf2_v4l2
> videodev              132913  4 bcm2835_v4l2,v4l2_common,videobuf2_core,videobuf2_v4l2
> media                  11573  1 videodev
> bnep                   10340  2
> hci_uart               17943  1
> btbcm                   5929  1 hci_uart
> bluetooth             326105  22 bnep,btbcm,hci_uart
> brcmfmac              186339  0
> brcmutil                5661  1 brcmfmac
> cfg80211              427855  1 brcmfmac
> snd_bcm2835            20447  1
> snd_pcm                75762  1 snd_bcm2835
> rfkill                 16037  4 cfg80211,bluetooth
> snd_timer              19288  1 snd_pcm
> snd                    51908  5 snd_bcm2835,snd_timer,snd_pcm
> bcm2835_gpiomem         3040  0
> bcm2835_wdt             3225  0
> uio_pdrv_genirq         3164  0
> uio                     8000  1 uio_pdrv_genirq
> i2c_dev                 5859  0
> fuse                   83653  3
> ipv6                  347466  49
> root@raspberrypi:~# ls /dev/
> autofs           full     loop4             net                 ram13   raw      tty11  tty22  tty33  tty44  tty55  tty9       vcs2   vcsa6
> block            fuse     loop5             network_latency     ram14   rfkill   tty12  tty23  tty34  tty45  tty56  ttyAMA0    vcs3   vcsa7
> btrfs-control    gpiomem  loop6             network_throughput  ram15   serial1  tty13  tty24  tty35  tty46  tty57  ttyprintk  vcs4   vcsm
> bus              hwrng    loop7             null                ram2    shm      tty14  tty25  tty36  tty47  tty58  uhid       vcs5   vhci
> cachefiles       initctl  loop-control      ppp                 ram3    snd      tty15  tty26  tty37  tty48  tty59  uinput     vcs6   watchdog
> char             input    mapper            ptmx                ram4    stderr   tty16  tty27  tty38  tty49  tty6   urandom    vcs7   watchdog0
> console          kmsg     mem               pts                 ram5    stdin    tty17  tty28  tty39  tty5   tty60  vc-cma     vcsa   xconsole
> cpu_dma_latency  log      memory_bandwidth  ram0                ram6    stdout   tty18  tty29  tty4   tty50  tty61  vchiq      vcsa1  zero
> cuse             loop0    mmcblk0           ram1                ram7    tty      tty19  tty3   tty40  tty51  tty62  vcio       vcsa2
> disk             loop1    mmcblk0p1         ram10               ram8    tty0     tty2   tty30  tty41  tty52  tty63  vc-mem     vcsa3
> fb0              loop2    mmcblk0p2         ram11               ram9    tty1     tty20  tty31  tty42  tty53  tty7   vcs        vcsa4
> fd               loop3    mqueue            ram12               random  tty10    tty21  tty32  tty43  tty54  tty8   vcs1       vcsa5

dontspammexyz, 2016-10-05:
I did connected the camera to display. Bad misstake from me.
Thanks for fast replys.
[image]

closed 2016-10-05 by dontspammexyz