motioneye-project/motioneyeos #545: Load balancing optimisation

gtr33m, 2016-10-06:
My camera system uses all network cameras and as I have added more cameras to a total of 3, which overloads the capacity of the Raspberry Pi so I have added an orangepi to balance the load.

I motion capture with a mask and stream video to android apps and the web from motioneye in 720p.  I would like to setup the system the most efficiently, and given the great flexibility of motioneye I can accomplish this in any number of ways.

I could set up each board individually, recording and streaming the cameras completely independently.

Or I could setup I board as the host adding the remote motioneye cameras and record and stream from the host.

Or I could setup some form of hybrid, with recording from each board to a shared resource but stream from one device for easier management.

Without testing each scenario, can anyone give me an idea which scenario would be most efficient, and an idea of how much overhead adding a remote motioneye camera adds? how much streaming adds? how much recording adds?

Ultimately I'll load balance manually until the system load is well balanced, but it's pointless if the remote processes add too much overhead.

Regards,

Mark

ekdrishta, 2016-10-06:
https://github.com/ccrisan/motioneyeos/wiki/Usage-Scenarios
I think it's already done.
I am just curious what you mean by "orangepi to balance the load" and how to did that.
I am planning to doing this same thing in my house.

gtr33m, 2016-10-06:
I'm not doing anything complicated like a pi cluster, just a combination like the usage scenarios.

I chose the orange pi pc for a number of reasons, but mainly because it is very cheap and yet powerful. I reasoned that it was more flexible to have the ability to go down to one SBC per camera if needed, rather that have everything on one more powerful SBC.

The orange pi pc is under $15 and the orange pi mini is $10 + a couple of dollars freight.  Add in an SD card and you are still under $20.

So far, with limited testing the orange pi can support 2 cameras at 720p with motion detection, recording and streaming.  Once I've got it all set up properly with ffmpeg compiled with x265 I'll post real world results.

ekdrishta, 2016-10-07:
orange pi supports motioneyeOS? while building from the source what changes should I make? I was planning to get 1920p with motion detection and stores in HDD(connected via USB). but without proper video compression standards video size would be gigantic.

gtr33m, 2016-10-07:
No, not yet.  I've asked the seller to send a sample board to the developer, but who knows if that will happen.  I'm running motioneye on armbian.  It's simple to setup, same instructions as raspbian.

I've confirmed that the Raspberry pi does not have h265 hardware support, which would explain your low framerates.  I can't confirm the h264 hardware support is built into the motioneyeos ffmpeg build, but I have found a ffmpeg build with h264 hardware support and h265 is a supported function of the Allwinner H3 which is the Orange Pi SOC chip.

Once a I get a working ffmpeg, I'll be able to test properly.  Software encoding will never work properly at such a high resolution on such low powered devices.

On another note, I think you are going to burn through USB hard drives unless you buy and enclosure and a PVR hard drive.  Video recording kills normal drives very quickly.

ekdrishta, 2016-10-07:
portable 2.5 inches USB drives give the 3-year warranty.
what PVR do to save the drive?
ekdrishta@gmail.com (can we talk there?) this is not anyway related to motioneyeOS. And I must use HDD to save such large data.

closed 2017-06-01 by bortek