motioneye-project/motioneyeos #546: Which is the most efficient codec for movie encoding?

gtr33m, 2016-10-06:
Based on a Raspberry pi 2, which codec is the most efficient?

ekdrishta, 2016-10-06:
I did little experiment on all the provided codec and I found this
1.mov and mkv give really large size video file 450-500 MB(480p, 5min continuous recording)
2.ogg and h265 freezes (if I know somehow how to overclock my pi this might not happen  @gtr33m   @ccrisan ) 
3.mpeg and mpegv2 works totally fine 40-50MB (480p, 5min continuous recording)
4.swf sucks

gtr33m, 2016-10-06:
Thanks.

I'll also test and post some results once I've got ffmpeg compiled properly

joshlukas, 2016-10-10:
Just a hint. Mov, ogg and mkv are not codecs. They all are container in which an encoded video can be saved.

x264, x265, dirac, mpeg-1, etc. are codecs which gives you the possibility to compress a video in their way. Mkv is "just" an container which holds the encoded data and gives an apropriate program the possibility to use some more options like, embeded subtitles, several audio streams, tags, etc.

For a rasperry pi to compress a high-res video with almost every single video codec available is just too much. It siply lacks of performance as it must encode and decode the video in realtime. Decoding (showing) a video is not a big problem for the raspi as it's done in hardware for the common codecs, but encoding a video is done in software due to missing hardware acceleration for that kind of task.

Next point is: What is efficiency? Compared to x264, x265 is more effictient for really high resolutions, but not quite as good for low-res videos if you have in mind that it takes more performance to achive the same result as you could have had done with mpeg-2.

bortek, 2017-02-23:
I am closing this as this is "a bit" :)  out of the scope of the project besides there has not been any discussion since few months back.

closed 2017-02-23 by bortek