motioneye-project/motioneyeos #547: Does motioneye support x265?

gtr33m, 2016-10-06:
My new board supports hardware x265, but my current ffmpeg does not.  I could, with difficulty, build ffmpeg with x265 support but it's a pointless exercise if motioneye does not support it.

ekdrishta, 2016-10-06:
motioneyeOS have HEVC/x265 option in GUI but in my raspberry pi 3 without any modification frames freezes even if I set 30fps frame rate(under video device and video streaming) it would seems like 4-5 fps. This same thing will happen with .ogg

ekdrishta, 2016-10-06:
if you can get smooth streaming in motioneyeOS with HEVC compression standard that really can make difference.

gtr33m, 2016-10-06:
Have you checked the compiled codec support of ffmpeg?

`ffmpeg -codecs | grep x265`

I am using motioneye on raspbian and armbian with the pre-compiled ffmpeg which does not have libx265 even though it does have HEVC support.  Without libx265 you are getting software encoding only which would explain the low frame rate.

Compiling ffmpeg is proving to be difficult, but I'm sure I could get it to work.  I just wasn't sure that motioneye would support it (as hardware encoding) and didn't want to persist with it if it didn't.

ekdrishta, 2016-10-07:
I am using motioneyeOS. and I think **yes**.

[image]

ekdrishta, 2016-10-07:
am I?

bortek, 2017-06-01:
Closing this issue due to inactivity. If stil interested please post your question in the forum page.

closed 2017-06-01 by bortek