motioneye-project/motioneyeos #548: there is no ARM a52 in xconfig

ekdrishta, 2016-10-06:
I want to use every bit of my raspberry pi 3 so what should I select while building from source scenario.
I really don't care if pi got heat up like hell I need superior performance.  
I don't know much about kernel compiling if someone can jog me through the process that would really help.
Thanks in advance

ccrisan, 2016-10-06:
I would wait for the next release of motion which addresses (among others) the problem of codecs, quality settings and compatibility with newer ffmpeg versions. Then it would be able to take advantage of the hardware accelerated h264 encoding available on the Raspberry PI.

P.S. what is ARM a52?

ekdrishta, 2016-10-07:
sorry typo error 
I was talking about pi3's ARMv8 cortex **a53**

ekdrishta, 2016-10-07:
using image file which is built from source uses all the threads of my raspberry pi3? if I use default settings,just follow the instructions, no changes at all.

bortek, 2017-06-01:
Closing this issue due to inactivity. If stil interested please post your question in the forum page.

closed 2017-06-01 by bortek