motioneye-project/motioneyeos #551: automatically timelaps every day

thundergreen, 2016-10-07:
Hey guzs,
absolutely great software what u created here.

Just an enhancement for the timelapse.

Would it be possible to integrate a schedule for automatically create timelaps every day with the poossibility to configure with the same options like actual.

would be nice. like this i could check in the evening what happened over the day.

Higly appreciate a resopnse

Greety

Thunder

thundergreen, 2016-11-02:
Hi all again. Is this feature not wished here?

Snille, 2016-12-15:
Hi! I'm also very interested in a timelapse function. It is already possible with motion. I have used when setting it up manually (for my 3D-printer). Would be very nice to be able to set it in the GUI as well! :)

ccrisan, 2016-12-16:
If you configure your motionEyeOS to take still pictures (snapshots) at regular intervals (e.g. 60 seconds), you can download a timelapse movie made of them from the media browser.

closed 2016-12-16 by ccrisan