motioneye-project/motioneyeos #553: bash script for action button

ekdrishta, 2016-10-09:
[image]
  
As you can see I added the action button on the top layer but there is no action while pressing.
I think I have to write something inside those file but I really have no idea.
If anyone has already written any of those those bash script please share or mail me at ekdrishta@gmail.com
These action buttons are for PTZ (I think) but I have not connected any motors yet and digital zoom and scroll inside of screen if more usefull because pi cams are 5 to 8 MP camera normal screens like mine are  720p is not enough.

ekdrishta, 2016-10-12:
@ccrisan @pauly @mikebanks  anyone knows??

crw, 2016-10-13:
The problem with this question, and why I suspect you will have difficulty receiving an answer, is that desired outcome of pressing those buttons is entirely in your hands. You can have them do anything you want. 

For instance, many moons ago I built [this pan/tilt swivel](http://www.instructables.com/id/Remote-Controlled-Pan-and-Tilt-Head/?ALLSTEPS). The control mechanism for the pan/tilt motors could be scripted with bash, and assigned to arbitrary buttons in motioneye. If you had built this exact same widget, and were using [my code](https://github.com/crw/pySwivel) to drive it, I could offer you a set of scripts that makes it move. 

The point being, everyone is going to have very specific requirements for their own setups, so without very specific information, no one can help you with your script request. 

As for having generic functionality to zoom the image and pan within the zoomed image, that is an interesting feature request but outside the scope of bash scripting and those buttons.

JohnnyCantTube, 2016-12-21:
@ekdrishta ;

In April, CCrisan moderator wrote just one good page on this;
https://github.com/ccrisan/motioneyeos/wiki/Action-Buttons
on this page, He wrote about the Action Keys you mention;
https://github.com/ccrisan/motioneye/wiki/Action-Buttons

putting the two together, it looks like you edit a specifically named Action Button file
using the NANO utility with a crafted BASH script.

I added some BASH shell script commands to "strobe" the left and right buttons
like 300+ milliseconds but searching for the _usleep_ and _sleep_ commands.

I hope this helps
Happy Holidays
- - -
(to elaborate my application for LEFT, RIGHT and LAMP ON)

I used/modified a small rotating Photo Cube plastic platform run by 4x AA
found for $5 in a curio-shop, as a "PAN" mechanism for a Driveway monitoring Cam.

 Four years ago, a Courier Truck entering my driveway to turn around
shaved off the passenger side Mirror on my truck while delivering to 
my Neighbor's residence across the road, 
and although my SMC Cam was 640x480 low crude resolution, 
I was fortunate that iSpy64 open source software (which was called something else then) 
captured the moment the Truck backed into the door mirror, 
and provided me with settlement evidence that I otherwise would have missed in absence.

I have modified the DC motor so that LEFT  and RIGHT
are "strobing" 2 GPIO pins to provide "PAN" to a RPi CSI connected 5MP camera.
That PAN table turns the RPi weather exterior case with camera and the Ethernet (PoE) cable.

Its a bit jerky but the PAN function works even if the outside temps drop to -30C.
I constructed cabling to use the Ethernet cable pins 7&8 to power the RPi as well.
The RPi board heats up the enclosure nicely, to keep the cam window defogged.

The LEFT_1, RIGHT_1, LAMP_ON , LAMP_OFF (IR LEDS in my build) are all I need from MotionEye. 
The RPi Camera 5MP is more than enough to identify (and Video log) license plates on my driveway.

I wanted to share this practical example and backstory
but I will also share hardware detail in a pending _Instructible_  or tutorial. 
I used an H-bridge of four MOSFETs (2xN-, 2xP-channel) 
to control a DC motor forward/reverse at 5V, but it takes two GPIO pins,
and some filtering/isolation of that +5V because a DC motor is "noisy".

I now realize that if I learn a bit more BASH code, I can build a dual Servo Pan&Tilt
using 9G servo motors(2x) and the Chinese Kits for the mechanism for about $20 cost
and right the PWM code to add UP and DOWN but for a driveway cam, 
its not really needed.. however, my Wife wants to know which Animal has been eating 
flower in the garden.

bortek, 2017-06-01:
Closing this issue due to inactivity. If stil interested please post your question in the forum page.

closed 2017-06-01 by bortek