motioneye-project/motioneyeos #554: ssh access

MBerger1, 2016-10-09:
Hi Crisan,

recently upgraded to 0915, and now am unable to ssh to the board...any suggestions on what to troubleshoot?

[image]

ccrisan, 2016-10-09:
Please attach your log files. The sshd service might have not been started for some reason.

MBerger1, 2016-10-09:
I bounced the boards, all good, ty sir and love motioneye!

ginoainsley, 2016-10-12:
I am also having issues ssh-ing into the the raspberry pi.  Even though the service settings is enabled, sshd does not start according to the boot.log.  It's a bit weird because I started enabling and disabling the various services and ONE time I rebooted and the sshd has actually started properly and I was able to ssh in.  So sshd CAN actually work...not sure why it is not starting.

I am using : motioneyeos-raspberrypi2-20160919

Attached are my logs.

@Mberger1 : what do you mean "I bounced the boards"?  

[logs.zip](https://github.com/ccrisan/motioneyeos/files/523236/logs.zip)

ginoainsley, 2016-10-12:
Update on my last post.  It is starting to look like the SSHd issue occurs with the default overclocking setting of (1000,500,500,2).  When I put the setting to None or Modest things seem ok.  I have a 16gb  SD card on the PI2.  Any other overclocking settings results in SSH not working

MBerger1, 2016-10-12:
gino - rebooted

tehXor, 2016-10-12:
sshd doesn't seem to run for me, too. So in the bootlog it doesn't show up in the start sequence and on restart shutting down sshd fails (probably because it's not running). Trying to ssh to motioneye anyway results in a timeout like expected, so sshd really doesn't seem to run.

I tired motioneyeos-raspberrypi2-20160919 and motioneyeos-raspberrypi2-20160915 both with the same problem.

I'm totally new to motioneye, do I have any other way of modifying its config files/underlying linux? Or can someone post a workaround? Simply switching services on and off and rebooting didn't do the trick for me :(

Thx in advance!

ginoainsley, 2016-10-12:
@tehXor 

Actually I doubt the problem is with sshd or the distro itself.  I would think the issue is the fact that the PI is overclocked - so for me, the combo of my PI2 and my 16g SDCard is likely unhappy at these higher frequencies.  Maybe I am just lucky anything works at all...not sure.

So it would be a good idea to have the default settings at "None" instead of any overclocked setting.

jibasig, 2016-12-17:
@ccrisan I have a similar issue with motioneyeos-raspberrypi3-20161212.img.gz on raspi 3. I have enabled ssh via web interface but sshd didn´t start at boot. Manually starting it from connected terminal worked.
Then I debugged the init sequence and had a look at /etc/init.d/S60sshd. There is a check for "has_net_conn" which seems to return false in my case. This is why any further code isn't executed anymore.
This seems to be a race condition with (in my case) the wifi setup.
I disabled the check and everything is working now as expected. The same workaround is successfull for ftp, too(anywhere where the "has_net_conn" check is done).
Could you fix the implementation of the check to just check "$os_networkless" (/etc/init.d/conf)?

hxpaul, 2016-12-21:
maybe fork the code and put in a pull request? That's how I did it

kutysam, 2016-12-22:
Tried multiple versions. 
Working ssh: 20160915
Anything above that, have to be started manually...

ccrisan, 2016-12-22:
@jibasig thanks for identifying the root cause here. I'll look for a fix.

kutysam, 2016-12-22:
Just have a hunch that it is regarding this file
https://github.com/ccrisan/motioneyeos/blob/7fe6ed1eb6ef2291b830c3711857595770f01f9d/board/common/overlay/usr/share/post-upgrade/20160828.sh

ccrisan, 2017-02-07:
@kutysam can you post here a boot log where the SSH service isn't started? I'm particularly interested in what the network service has to say.

ccrisan, 2017-02-09:
@kutysam https://github.com/ccrisan/motioneyeos/commit/28cb7634c7cdcbee40019aa62aa1183ae25d4899 will fix your particular problem with SSH not being started, but I fear that the root cause of your problems is in fact the overclocking.

closed 2017-02-09 by ccrisan