motioneye-project/motioneyeos #556: Where is the video log?

vyking100, 2016-10-10:
Hi - Can some help me please. I have Motioneye up and running fine via the latest iamage file. However I cannot see any video logs to view the saved files.  These are the logs as shown under Expert settings.
Log Files: 	
motion.log
motioneye.log
messages.log
boot.log
dmesg.log
streameye.log	
raspimjpeg.log
They all show text only.
I'm thinking its something to do with the config file but when trying to veiw the Pi via Putty I am denied access.
Any help would be really appreciated.

crw, 2016-10-13:
First, my instructions assume you have turned General Settings > Advanced Settings on. 

1. ssh access: you can login via ssh if you have Services > Enable SSH Server set to "on" (default). If you set a password for admin, use that password to login. I do not use putty but a typical ssh command would look like: `ssh admin@ip-address-of-motioneyeos`

2. Video log. I am not sure what you mean by "video log." I am not personally familiar with that file. If you have Samba enabled (Services > Enable Samba Server), you should be able to use Windows networking to see two shares being served by your motioneyeos device: sdcard and storage. If you look at the setting for "File Storage > Root Directory", this tells you where the motion captured files are stored. 

Note that Movies need to be turned on and Motion Detection needs to be turned on to record movies in the first place.

vyking100, 2016-10-13:
Thank you for replying - I have ssh access thank you. My problem it that I do not know how to view the stored video. 
File Storage tell me:
Sorage Device = Custom Path
Root Directory = /data/output
the disk usage is showing 1% of theavailable space is used so something must be stored.
Movies is turned on as is Motion Detection.
I can get to the root diectory on the Pi but not sure of the correct syntax to get to the file I need.
Many thanks again

crw, 2016-10-13:
Cool. When you are logged in via ssh, what does `ls /data/output` show you? If you haven't changed any of the default settings, you should have a folder structure that looks something like this:

    [root@Cambot01 ~]# ls /data/output/Camera1/
    2016-10-08/ 2016-10-09/

Within each of those datestamp folders are a bunch of video files and thumbnails. If you do not see any of these folders, you are not getting video recorded for some reason.

Note that for some reason, the `/data/output` folder is shared via Samba as `sdcard`. I am not really sure why since everything is on the SD card, nor do I understand why there is a separate `storage` share. Anyway I am guessing you are on Windows, if you "Map Network Drive..." using the ip address and "sdcard" share, you should be able to navigate the file system in Windows File Manager. Let me know how this goes for you.

vyking100, 2016-10-13:
Graig - Thank you so much for your imput. I have just - finally - sorted 
it out. It's one of those 'can't see the wood for the trees'. Its taken 
me two weeks but from what I can now see was worth it.

Once again many thanks.

Richard

closed 2017-06-01 by bortek