motioneye-project/motioneyeos #560: SSH hangs after password on Raspberry PI 3 with internal Wifi Adapter

ggramlich, 2016-10-16:
As I mentioned in #181, I was unable to ssh into my fresh installation on Raspberry PI 3, when using the IP address assigned to the internal wifi adapter. (SSH on cable LAN was not a problem.)
The login hung after entering a correct password.

This seems to be a common problem with RPI3. I found this thread
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=138631&start=100
And followed the suggestion that came up at the end of page 4 of the thread and was confirmed multiple times on page 5.

Added the line

`IPQoS cs0 cs0`

to /etc/sshd_config worked for me.

ccrisan, 2016-10-16:
Thanks, I'll look into it.

ggramlich, 2016-10-16:
See also
https://expresshosting.net/ssh-hanging-authentication/

robcop2000, 2016-10-17:
Same problem I can enter with passowrd in Motioneye true the web interface, but not with terminal

ccrisan, 2016-10-26:
Fixed with d2dfeb9481fc625d25bfc4b97954a5ae4aae61b9.

closed 2016-10-26 by ccrisan