motioneye-project/motioneyeos #561: Motion detection mask feedback

jimuhall, 2016-10-16:
Thanks again for implementing the motion detection mask in issue #483 .  Some feedback - it does appear to limit the area of detection, but maybe not as I would expect.  Some questions and comments:

1. Is it possible to have multiple open areas in the mask as I am showing in mask.png?  I would like to detect motion only in these two areas.
2. I am getting unexpected motion as shown in Image0.jpg and Image1.jpg.  The pink box is drawn larger than my unmasked areas - does this mean that it is still detecting the tree as it moves in the wind, outside of my unmasked area?  
3. I have my frame change threshold set very low (1%), which seems necessary to see automobiles or foot traffic.  If I go as high as 2% I miss foot traffic.  Maybe the shadows from trees are still causing a detection event?

Thank you!
 
[image]
[image]
[image]

xxllo, 2016-10-16:
Super software. Thank you very much

ccrisan, 2016-10-17:
Thanks for your feedback @jimuhall. You're the first one to actually provide some feedback about this new feature. Unfortunately I couldn't possibly help you with an answer. This is strictly a motion internal thing and the best place to find an answer on this matter would be the [official motion project's page](https://github.com/Motion-Project/motion).

I personally ran some not-so-thorough tests and it wouldn't detect motion in areas masked out. It appeared to me that it worked as expected.

There's one more thing. Could you please check and make sure the generated mask image (usually `mask_1.pgm`) has the same dimensions as your stream? There's some work in this sense performed  by motionEye (along with a patch to motion) and there might be some bugs.

Schmello, 2016-10-17:
HI... i have install Motion Eye on Linux Mint..
How can i get this feature?..
I try to update Motion Eye... 
It is up to date..
Current Version 	0.33.4

ektus, 2016-10-22:
I've come around to installing the new version of motioneye today, using the "pip update motioneye" command as recommended. I'm on Raspian Wheezy, and the newly installed version reports in as 0.34rc1

The edit mask feature is working as desired, very good. After editing /usr/local/lib/python2.7/dist-packages/motioneye/utils.py and inserting the correct picture format of my camera, that is. 
I'm running on the preview stream of a RTSP camera, which is 640x360. Most of the evening was spent searching for that file, though :-( This comment is mainly for storing the above path online, that's where pip installs motioneye.

I had to copy over my existing mask file from the /home/pi/motion-mrdave directory to the /etc/motioneye directory and rename it to mask_2.pgm, then it was recognized by motioneye and I was able to edit in from within the UI.

After the update, the extra motion commands field is empty.

ccrisan, 2016-10-22:
@ektus what was your problem, more precisely? What did you have to edit in your utils.py file? What did *not* work out-of-the-box?

ektus, 2016-10-22:
utils.py: Nothing mask related. Camera resolution is not in the "common resolutions" list and custom resolutions are still on the "ToDo". Therefore I had to edit utils.py to add my resolution (640x360) so that it won't be overwritten when changing settings through the UI. I've done it before, but couldn't remember the file location :-(

Not "out-of-the-box": mask filename and location from the extra motion commands field was not picked up by the new version, but the contents of that field was deleted. I've had some "file not found" error messages in the console that went away after relocating the mask file to the motioneye directory and renaming it.

ccrisan, 2016-10-24:
Ah i see. Indeed the resolution problem is still there :) The mask file location is important as it's being interpreted by both motion and motionEye. Anyway thanks for your feedback!

martyouel, 2016-10-26:
Guys,

I tried upgrading my version to get the 0.34RC1 and try the motion mask.
I'm running MotionEye on Raspbian, so I ran :

sudo pip install motioneye --upgrade                
sudo systemctl restart motioneye

But i'm still stuck at 0.33.4

any clue ?

thanks

ccrisan, 2016-10-26:
What does `sudo pip install --upgrade motioneye` say? I just pushed 0.34.1 today, by the way.

martyouel, 2016-10-26:
Calin, thanks a lot !
I'm running the 0.34.1 now !
continue your good work !

martyouel, 2016-10-27:
Not sure if it's ok that I continue this thread with question related to motion mask....if not i'll start another.

When I try to "Edit mask" for a mask type "Editable" I get : Cannot edit the mask without a valid camera image! 
Not sure what's the problem here.

I have no problem to activate the "Smart" mask but I don't really know if this work.

thanks,

ccrisan, 2016-10-27:
The message you receive is normally due to missing live stream from the camera. What camera type do use and are you sure the camera stream is working at the moment when you try to edit the mask?

martyouel, 2016-10-27:
I have a trendnet ip312w...i know it's a old camera.
I am connected to the cam through this address : http://ip/cgi/mjpg/
The stream is working great when I try to edit the mask.

thanks

ccrisan, 2016-10-27:
@martyouel would you be able to temporarily give me access to your web interface so that I can investigate the problem?

martyouel, 2016-10-27:
I just sent you 2 emails with all the information.
Hope it will help you.
Waiting feedbacks
thanks

ccrisan, 2016-10-27:
I found the root cause of the problem and I'll come up with a fix. Thanks for your time.

martyouel, 2016-11-03:
Hi guys,

I think I have the same situation than ektus had.

I get a lot of false motion events and i'm pretty sure those false events are coming from my window.

I then built a mask to mask those sources of light but it seems that I still receive motion from those areas.

[image]

Here's my mask settings 

[image]

I don,t really understand what ektus changed.

Somebody can help me figure it out ?

ccrisan, 2016-11-03:
Could you please manually add the option "despeckle_filter EedDl" to your thread config (or via Extra Motion Options)? It is said to improve false detections a bit.

martyouel, 2016-11-04:
Thanks Calin.
I tried this yesterday, I thought that it was working better but as it was a dark day I didn't get a lot of light change. Today, it's a clear day, and I began again to receive more false motion event.

My main question is still there, is it normal that Motion seems to detect motion even if I putted a mask on my door and big window ?

Any clue ?
thanks guys

ektus, 2016-11-04:
There may be shadows on the floor. Even clouds or smoke from a chimney can cause motion detection. But the very high change count in the upper right corner indicates some other problem. You could toy around with the light switch detection setting, trying to cut off changes that effect a large area of the image.

martyouel, 2016-11-04:
Ok, i'll try work with that.

But do you know why I get motion alert in a masked area ?
From my side it looks like the mask do not work in my case.
Take a look of this alert :

[image]

[image]

is it "normal" ?

thanks

ektus, 2016-11-04:
Does the mask image size match the video resolution? Are there any mask-related errors in /var/log/motion.log? If the mask was working, detection inside the masked-off window should be impossible.
Yet again, try a lower light switch setting (like, 5 to 10%).

martyouel, 2016-11-04:
ohhh
thanks ektus, i'm pretty new in that world and didn't know about the motion log.

[1] [ALR] [NET] netcam_handler_loop: Camera handler thread [3] started
[1] [ERR] [ALL] get_pgm: Wrong image size 512x384 should be 640x480: Operation now in progress
[1] [ERR] [ALL] motion_init: Failed to read mask image. Mask feature disabled.

Indeed I have a problem with my mask.

How can I validate if "my mask image size" match my video resolution ?
Sorry for the newb question.

Thanks,

martyouel, 2016-11-04:
Ok, ignore my last post.

Found where to change it.
Then I changed it.

The "new error" is now :

[1] [ERR] [ALL] get_pgm: Failed reading image data from pgm file: Operation now in progress

ccrisan, 2016-11-04:
What version of motionEyeOS is this?

martyouel, 2016-11-04:
0.34.1

ccrisan, 2016-11-04:
That's motionEye. I thought the issue was about motionEyeOS.

You're probably using motion 3.2 or 3.4. These versions do require the mask to have the exact same size as the stream, indded, as @ektus pointed out. I provided a patch to motion that automatically resizes the mask according to the stream requirements. It has been included in 4.0.

If you were using motionEyeOS, the motion version included with motionEyeOS also has this patch applied.

martyouel, 2016-11-04:
Ok sorry, i'm not running on motioneyeos but on raspbian.

i'll take a look into upgrading motion.

thanks

martyouel, 2016-11-04:
Well, I upgraded to motion 4.0.1 and I get the same behavior.

1. I edit the mask in motioneye
2. I check the motion.log and I have an error get_pgm: Wrong image size 512x384 should be 640x480: Operation now in progress
3. I manually edit mask_1.pgm and change 512 384 by 640 480
4. I restart motioneye sudo systemctl restart motioneye
5. I check the motion.log and I have now get_pgm: Failed reading image data from pgm file: Operation now in progress

any clue ?
thanks

ccrisan, 2016-11-04:
Messages are different now: it no longer says "Mask feature disabled". As for the other error that you keep seeing (Operation now in progress), you'll need to check that with motion developers. This has nothing to do with motionEye.

ccrisan, 2016-11-13:
I'll close this issue as there's nothing that motionEye can do about it.

closed 2017-06-01 by bortek

Mikgabo, 2018-04-01:
Good day,

Has anyone had issues with the mask moving around?  Every time I go back to settings the mask has moved on the display.

Thanks,
Mike