motioneye-project/motioneyeos #564: Lag / Video stuttering

tehXor, 2016-10-19:
I have attached a cam with an RTSP stream (1280x720 and TCP currently) to motionEyeOS. When I open the same stream via VLC the video shows in good quality (well at least as good as this cam gets) and completely fluid. However in motionEye the video has a similar quality but is by far not fluid. So the video seems to lag / stutter. I have maxed out all FPS options but that doesn't seem to help. When I connect to the stream provided by motionEye (http://motion.eye:8081) I get the same stuttering.

Considering this is for surveillance purpose and not movie recording a little stuttering would be okay, however unfortunately due to the stuttering fast movements are sometimes not shown in motionEye while they are clearly visible in the original stream.

Is there anything I can do to improve this?

bortek, 2017-06-01:
no

closed 2017-06-01 by bortek

tehXor, 2017-06-01:
A pity - would you mind sharing some insights? Why are so many frames dropped? And there seem to be other projects based on WebRTC stream interpretation and manipulation which don't have these problems - unfortunately they don't cover the surveillance use case (on small ARM boards as a OOB solution) as good as motioneyeos currently would do. It's been a while since I had to do more with WebRTC but stream manipulation seemed very straight forward and search for WebRTC motion detection nowadays brings up a lot so maybe it's worth considering it...

Tejeev, 2020-09-07:
@tehXor did you find the RC of your lag?