motioneye-project/motioneyeos #566: Feature Request: Simple Motion Detection trigger to the external world

BeerStein, 2016-10-23:
Thanks for sharing this great piece of SW.  

Something that would be very helpful would be to be able to configure a GPIO output to be triggered when motion is detected. This will be useful for triggering a warning buzzer, turning on a light etc.

martyouel, 2016-10-24:
You can just build a python script to do what you want...and then, configure motioneye to execute your script on motion (Motions notification / Run A Command

BeerStein, 2016-10-25:
Thanks martyouel, that was obviously a NOOB question, which is what I am.   I can log into MotionEyeOS from my PC using Putty which is probably the first step. Now to find a simple "Hello World" sample of how it is done.

BeerStein, 2016-10-27:
I am missing a something somewhere?

Based on the you-tube tutorial uploaded by Chris Dahms I created a simple `my_blink.py` in the data directory as per the Push Notifications instructions by MakerZone

The script is to turn an LED ON for half a second and then turn it OFF 
 
```
# my_blink.py

import RPi.GPIO as GPIO
import time

GPIO.setmode(GPIO.BCM)      # use GPIO pin numbering

led_gpio_pin = 18
GPIO.setup(led_gpio_pin, GPIO.OUT)
GPIO.output(led_gpio_pin, GPIO.HIGH)
time.sleep(0.5)
GPIO.output(led_gpio_pin, GPIO.LOW)

```
using the `chmod` command it was made executable.

If I do a `python /data/my_blink.py` the LED will turn on for 0.5 sec and turn of as expected.

I then entered the following into the MotioneyeOS Command window
 `/data/my_blink.py` 
and enabled the Run A Command. 

Hit the `Apply` button

Moved my hand over the FOW to create an event, A Video clip of the motion event is saved but no blinking LED.

Also tried restarting the RaspberryPi with the above settings in place - no luck.  

I am sure this is a NOOB mistake. Any pointers will be appreciated?

ccrisan, 2016-10-27:
How about adding python in front of the command, as you did when you tested
it manually

BeerStein, 2016-10-27:
Yup, that did it!  Thanks!

closed 2016-10-28 by ccrisan