motioneye-project/motioneyeos #567: expose a local image url for telegram/slack integration

mpender, 2016-10-24:
nice app first off,

is there a nice way to say curl a local url like : http://192.168.0.x/picture/1/current/ so I can mangle together a webhook to say telegram? 

When i try to do something like : 

wget -O pic.jpg http://192.168.0.x/picture/1/current/?_username=admin&_signature=some_signature

curl -s -X POST "https://api.telegram.org/bot_token/sendMessage?chat_id=id/sendPhoto" -F chat_id=id -F photo="@pic.jpg"

I get a 403 on the wget as I use authentication (basic) on the streaming URL.

Maybe I could add a front end nginx proxy to add some authentication and leave motioneye more open? 
 
thoughts?

mpender, 2016-11-03:
suppose I better fork this project and see what way its actually configured, nevermind

bortek, 2017-06-01:
Closing this issue due to inactivity. If stil interested please post your question in the forum page.

closed 2017-06-01 by bortek

shahonseven, 2018-02-25:
After many tries, I manages to send notification via telegram

```curl -s -X POST "https://api.telegram.org/bot<token>/sendPhoto" -F chat_id=<chat_id> -F photo=@/home/ftp/sdcard/Camera1/$(date '+%Y-%m-%d')/`ls -utlr /home/ftp/sdcard/Camera1/$(date '+%Y-%m-%d') | grep \.jpg | tail -n1 | awk '{print $NF}'` ```