motioneye-project/motioneyeos #568: Video streaming to Domoticz

rafi1975, 2016-10-25:
Hi,

I know that's not the Domoticz forum but someone may help. I asked their forum and no one could fix that.
I'm trying to stream to that software and part of it fails. The way you configure it is just to provide IP of the motioneye box with port and there's the last bit that must be wrong. It's called imageURL.
When I try to add the camera providing its IP and port and even user I can the video on the preview but once added there's no output. I can see the following error in motioneye.log:

WARNING: 403 GET /login/?_=1477322137351&_username=admin&_signature=70476c13fbbebf1028feaf6a0a80d0b5225d3446 (192.168.1.218) 4.74ms

[image]

On their Wiki there are some examples adding another vendor's cameras e.g.:

To setup a Axis 221/211 enter the following:
IP address: the address of the IP camera (e.g. 192.168.1.111)
Port: the port of the IP Camera (e.g. 80)
Username: <username of the IP cam>
Password: <password of the IP cam>
ImageURL: axis-cgi/jpg/image.cgi

Any one any idea?

Thanks,

Rafal

ccrisan, 2016-10-25:
The signature-based authentication mechanism used by motionEye is probably not implemented in Domoticz. This is not going to work unless either they implement it or I implement the good old HTTP authentication scheme in motionEye.

rafi1975, 2016-10-26:
Thanks for the reply. It means I have to wait either for you or them.
Patience is needed then.

bortek, 2017-06-01:
Closing this issue due to inactivity. If stil interested please post your question in the forum page.

closed 2017-06-01 by bortek