motioneye-project/motioneyeos #569: motion boxes becoming current?

fundean, 2016-10-25:
Hey there CCrisian
Any chance you know when you will make the current version the one with the masks?
it works amazing but I don't want to pull the cameras down to update... Hoping to be able to just use the apps update feature and the current one doesnt have the boxes yet...

Thank you again its awesome
Dean

ccrisan, 2016-10-26:
There still are a few minor issues to be dealt with. I believe the stable release will be out within 2 weeks or so.

scoobyd00, 2016-11-09:
Looking forward to this feature on the stable release :)

ccrisan, 2016-11-10:
If by "motion boxes" you guys understand mask support, then it's gone stable already: https://github.com/ccrisan/motioneyeos/releases/tag/20161027

closed 2016-11-10 by ccrisan