motioneye-project/motioneyeos #575: Email Notification Fail (V20161027)

stllunatic, 2016-10-29:
I updated to version 20161027 and the email notification system fails. 

Notification email failed: [Errno -3] Temporary failure in name resolution!

ccrisan, 2016-10-29:
How about your dns settings?

stllunatic, 2016-10-31:
I also added a router to my configuration (prior to this everything was working). I've entered the DNS Server that matches my router. I am also defaulted to the date time stamp of 1969. I tried alternating the Date Method between NTP and HTTP (Google and Bing) without any luck.

ccrisan, 2016-10-31:
Do you have a static network configuration for your motionEyeOS unit? What is your DNS server set to?

stllunatic, 2016-11-02:
It appears to be a problem with the compatibility of my modem and router. I'm using a Motorola SBG6580 and a Netgear AC2600. Because the modem is also a router it is causing the error with the DNS server and the IP address isn't "passing through" correctly, even when the wifi is disabled on the modem. Probably going to purchase a new modem that is not "wireless". Sorry for adding an unrelated issue. Maybe it will help someone in the future with a similar problem.

closed 2016-11-02 by stllunatic

stllunatic, 2016-11-09:
Another lesson learned here was that my DNS settings were not right. When the DNS is incorrect, notifications and your time are wrong. I found my DNS within my router settings. This number didn't work, but when I went into a PC and went to run, command prompt, and ipconfig /all ...then the correct DNS, which worked for the Rpi was listed.