motioneye-project/motioneyeos #576: Dropbox - Location? - Motioneye uploader?

Jimbojay, 2016-10-30:
Hi There,

i tried to get the dropbox storage up and running but I am experiencing some questions/troubles.

1. What has to be filled in at the 'Location' submenu at the 'File Storage' menu
2. If i press 'obtain key' it redirects me to dropbox where I authorize. But in the end it says: **"Enter this code into motionEye Uploader to finish the process."** where can I find this Uploader?

Kind regards,
Imre Dekker

ccrisan, 2016-10-31:
1. Whatever folder path you want your files to be stored into, in your dropbox account.
2. Use the Authorization Key field in motionEye.

Jimbojay, 2016-10-31:
Hi ccrisan,

thanks for your swift reply! 

1) created a new folder 'Home' in the dropbox directory
2) Filled in 'Home' as the location-submenu of motioneyeos
3) pressed obtain key and walked through the process
4) copied the code to authorization key
5) Test -->succesfull
6) because the key dissappeard pasted it again
7) pressed apply
8) recorded video message via motion detection --> video is in the library BUT not in dropbox.

Any suggestions?

Kind regards

ccrisan, 2016-10-31:
Do NOT paste the key again. It's a one-time key used by the test procedure, or by the first upload. It will be invalid if used a second time.

closed 2017-06-01 by bortek