motioneye-project/motioneyeos #583: Unable to open video device

pjockey, 2016-11-04:
Getting this error. The system was updated to, but no change. I noticed this last week.

I'm sure this has been reported be for, but I could not find any threads.

[Lof_files.zip](https://github.com/ccrisan/motioneyeos/files/572450/Lof_files.zip)

ccrisan, 2016-11-04:
It says:

    Failed to open video device /dev/video0: No such file or directory

This means that the system does not detect your camera device. Make sure the camera is correctly connected and functional.

pjockey, 2016-11-07:
Thanks Calin, got it working.Scott                                                                                                                                                                                                                                                                                                                                        Sent from my BlackBerry 10 smartphone on the TELUS network.                                                                                                                                                                                                                From: Calin CrisanSent: Friday, November 4, 2016 13:20To: ccrisan/motioneyeosReply To: ccrisan/motioneyeosCc: S Milliken; AuthorSubject: Re: [ccrisan/motioneyeos] Unable to open video device (#583)It says:

Failed to open video device /dev/video0: No such file or directory

This means that the system does not detect your camera device. Make sure the camera is correctly connected and functional.

bortek, 2017-06-01:
Closing this issue due to inactivity. If stil interested please post your question in the forum page.

closed 2017-06-01 by bortek