motioneye-project/motioneyeos #585: Edimax EW-7811Un no longer being detected after upgrade to 20161027

flyingclimber, 2016-11-06:
Looks like my WiFi dongle (EW-7811Un) isn't being detected on bootup after the 20161027 update. 

Is there any way to get it to stop rebooting automatically so that I could triage what is going wrong?

thanks

flyingclimber, 2016-11-07:
I'll try manually disconnecting the WiFi dongle and just using wired ethernet this week. Hopefully that will get me some better debugging info.

flyingclimber, 2016-11-11:
Seems to still be rebooting even without the adaptor. Guessing it's not trying to get a ethernet DHCP lease. Going to re-image.

flyingclimber, 2016-11-12:
Pretty sure this is just a power issue. Can't power the camera+infrared+wifi all at once with a 2A plug. Closing out.

closed 2016-11-12 by flyingclimber

ccrisan, 2016-11-12:
Thanks for keeping us posted. Should the problem occur again, please reopen the issue.