motioneye-project/motioneyeos #587: Action Buttons problem

HannesCr, 2016-11-07:
I have created an Action button but get the following error in the log-
2016-11-07 13:00:29: [motioneye]    ERROR: 500 POST /action/1/alarm_on/?_=1478516428633&_username=admin&_signature=32f33b652a1d7a109f3305b60050ebdba9c420b0 (169.0.218.64) 23.99ms

My script-
#!/bib/bash

GPIO=18
test -e /sys/class/gpio/gpio$GPIO || echo $GPIO > /sys/class/gpio/export
echo out > /sys/class/gpio/gpio$GPIO/direction
echo 1 > /sys/class/gpio/gpio$GPIO/value

ccrisan, 2016-11-07:
Make sure your script is executable and it starts with `#!/bin/bash`, not `#!/bib/bash`.

HannesCr, 2016-11-07:
Can't believe I have not seen it! Sorry for overlooking this, ID10T error on my side!

closed 2016-11-07 by ccrisan