motioneye-project/motioneyeos #588: Running bash script as custom command

navigonic, 2016-11-07:
Seems like a stupid question, but is it possible to run a bash file after a set of media is created? I've tried to experiment what syntax the "Run A Command"-box is designated for, aside for short commands seperated with semicolons:

None of these work so far:
1. sh link/to/bash/file
2. "sh link/to/bash/file"
3. link/to/bash/file

ccrisan, 2016-11-07:
How about simply `/path/to/bash/file.sh`, after you have made the script executable and added the `#!/bin/bash` line at the beginning of the file?

codecvlt, 2016-12-30:
ccrisan's solution to add #!/bin/bash is working for me, but if i include any bash parameters after my script it does not work. e.g /home/user/script.sh VAR1;

ccrisan, 2016-12-30:
@codecvlt command line arguments should be transferred from the Run A Command box to the script itself without any problem. What does your Run A Command box currently have set and how does your script look like?

codecvlt, 2016-12-30:
so initially, i had tried to put ```curl -X POST --data-urlencode "payload={\"channel\": \"#motioneye\", \"username\": \"webhookbot\", \"text\": \"Motion detected in LIVING ROOM\", \"icon_emoji\": \":ghost:\"}" https://hooks.slack.com/services/<mywebhookdetails>``` into the command field, because I couldn't get the webhook option to work. But when i'd go to save that, it would deselect the command toggle, and select the webhook toggle with my webhook url prepopulated as a GET request. This would not work, since slack requires a POST with a specific json payload (which i couldn't figure out how to set in the web UI). So then i put that command into a script, which i thought was working but found out it wasn't. I found out that I made a mistake in assuming that my script would be executed from the webhost (different linux box) than my remote motioneye camera (raspberry pie).

I've logged into the remote raspberry pie, and could see the error in my logs saying it couldn't find my script. I've since copied it over and updated my config, and now it works.

ccrisan, 2016-12-31:
I'll close the issue then.

closed 2016-12-31 by ccrisan