motioneye-project/motioneyeos #589: fails to build  -

rabarar, 2016-11-08:
Looks like there's a missing dependency in the Kconfig for the driver/media/i2c/ov2659.c

it should be:
```
depends on MEDIA_CAMERA_SUPPORT && VIDEO_V4L2_SUBDEV_API
```
rather than
```
depends on MEDIA_CAMERA_SUPPORT
```

see:
https://patchwork.kernel.org/patch/6436761/

ccrisan, 2016-11-08:
What kernel, what board, what version?

rabarar, 2016-11-08:
i attempted to build the motionpie repo for raspberry3 - cloned it and 

make raspberry3_defconfig
make

ccrisan, 2016-11-08:
1. Don't use motionPie. Use motionEyeOS instead.
2. Follow [these](https://github.com/ccrisan/motioneyeos/wiki/Building-From-Source) instructions to build your OS.

rabarar, 2016-11-08:
> Don't use motionPie. Use motionEyeOS instead.
> Follow these instructions to build your OS.

Cool - thanks!

closed 2017-06-01 by bortek