motioneye-project/motioneyeos #590: Stuck on rainbow screen

HaniKali, 2016-11-09:
Hi i have rpi 2 b+ have tried writing on serveral different memory cards but doesnt go past rainbow screen on boot.

ccrisan, 2016-11-09:
https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#system-wont-boot

chenselman, 2016-12-08:
The rainbow screen happens for me if I'm using the wrong image for the device.  I had that happen to me when I tried to use the rasppi 2 image on a rasppi 3. Use this one motioneyeos-raspberrypi2-20161125.img.gz  for raspberry pi 2. https://github.com/ccrisan/motioneyeos/releases
hope this helps!

toxeia, 2016-12-16:
Unfortunately I'm having the same issue. As far as troubleshooting, I've done the following:

1. Tested different (4) SD cards
2. Tested different distros on those SD cards
3. Disconnect everything but HDMI and SD card
4. tried a different rasppi (B1 and B2)

Am I missing something? These are my steps to getting the image into the RaspPi (Windows)

1. Use 7zip to extract *.img from *.img.gz
2. Use DiskPart to clean/remove partitions from SD card
3. Use SDFormatter to format SD card
4. Use Win32DiskImager to write *.img
5. Safely remove/eject SD card
6. Insert SD card into RaspPi B
7. Boot - rainbow screen forever.

Before submitting this I downloaded the GnuWin gzip utility for Windows and gave that a shot thinking possibly 7zip extraction of gz was weird. No change from the norm unfortunately.

ccrisan, 2016-12-16:
@toxeia, 

1. What version is written on your board?
2. What exact motionEyeOS image file have you downloaded?

toxeia, 2016-12-17:
The two boards I'm trying to use are "Raspberry Pi Model B+ V1 2" and "Raspberry Pi Model B V1 1." 

I've been using image files "motioneyeos-raspberrypi2-xxxxxxxx.img" and "motioneyeos-raspberrypi-xxxxxxxx.img" Releases for the dates: 20151011, 20161212, 20161027, 20161011.

ccrisan, 2016-12-17:
Then motioneyeos-raspberrypi-xxxxxxxx.img should work for you. Are you sure you have written the extracted img to the SD card? Do you by any chance have any other Linux machine around so you could see what's on the SD card after having written the image?

toxeia, 2016-12-17:
I was going to try writing the SD card from Linux today actually. I'll let you know what the results are from that.

toxeia, 2016-12-18:
Okay, I got around to writing the image from Linux and it works fine. I gave it a shot from my Windows laptop as well just to see if there was something wrong with the software or interface on my desktop, but I had the same issue. I'm not sure why I can't successfully write to the card from Windows.

vaderj, 2017-03-18:
I am having the same issue - When I write motioneyeos-raspberrypi2-20170212.img to a SD card with Win32DiskImager, my Pi2 never gets past the rainbow screen. 

Using the exact same card and hardware, I can write 2017-03-02-raspbian-jessie-lite and 2017-03-02-raspbian-jessie and the same pi boots up fine.

ccrisan, 2017-03-18:
What is the exact model as written on your board?

vaderj, 2017-03-18:
My mistake - This is actually a B+ board.  After using the image for the original pi, it boots up great!  Thanks for the help!

closed 2017-03-18 by ccrisan