motioneye-project/motioneyeos #591: Issue with motioneyes lastest verion

moonguar, 2016-11-11:
I have the lastest version of motioneyeOS installed on a RPI3 and have motioneyes installed on another RPI3.  Everything seems to be working great but I am experiencing one problem.  After I add the camera from the motioneye system into the motioneyeOS system GUI, the camera appears and is working fine but when I try to change a parameter on the remote camera from the motioneyeOS GUI, I am getting an internal server error when I try to "apply" the change.  Here is a copy of the error messages in the log:  

016-11-11 06:36:14: [motioneye]    DEBUG: setting config for remote camera 1 on http://192.168.1.7:8765/config/1
2016-11-11 06:36:14: [motioneye]    ERROR: failed to set config for remote camera 1 on http://192.168.1.7:8765/config/1: internal server error
2016-11-11 06:36:14: [motioneye]    DEBUG: 200 POST /config/0/set/?_=1478864175099&_username=admin&_signature=0e089a1750ebba7c70b15cb2d4e723db11b60a93 (192.168.1.7) 73.21ms
2016-11-11 06:36:17: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 2.87ms
2016-11-11 06:36:22: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 2.87ms
2016-11-11 06:36:23: [motioneye]    DEBUG: running garbage collector for mjpg clients...
2016-11-11 06:36:25: [motioneye]    DEBUG: setting config for camera 2...
2016-11-11 06:36:25: [motioneye]    DEBUG: reading camera config from /data/etc/thread-2.conf...
2016-11-11 06:36:25: [motioneye]    DEBUG: writing camera config to /data/etc/thread-2.conf...
2016-11-11 06:36:25: [motioneye]    DEBUG: setting config for remote camera 1 on http://192.168.1.7:8765/config/1
2016-11-11 06:36:25: [motioneye]    ERROR: failed to set config for remote camera 1 on http://192.168.1.7:8765/config/1: internal server error
2016-11-11 06:36:25: [motioneye]    DEBUG: 200 POST /config/0/set/?_=1478864186240&_username=admin&_signature=12e1f29f3ecd6a5144ba2db52436cb5bd3f9c3ca (192.168.1.7) 142.00ms

Any ideas?

ccrisan, 2016-11-11:
More useful details appear in the remote motionEye log file.

moonguar, 2016-11-11:
Here is the output of syslog on the remote system

Nov 11 07:26:37 raspberrypi meyectl[451]: DEBUG: setting config for camera 1...
Nov 11 07:26:37 raspberrypi meyectl[451]: ERROR: 'despeckle_filter'
Nov 11 07:26:37 raspberrypi meyectl[451]: Traceback (most recent call last):
Nov 11 07:26:37 raspberrypi meyectl[451]: File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1425, in _stack_context_handle_exception
Nov 11 07:26:37 raspberrypi meyectl[451]: raise_exc_info((type, value, traceback))
Nov 11 07:26:37 raspberrypi meyectl[451]: File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1627, in wrapper
Nov 11 07:26:37 raspberrypi meyectl[451]: result = method(self, *args, **kwargs)
Nov 11 07:26:37 raspberrypi meyectl[451]: File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 246, in post
Nov 11 07:26:37 raspberrypi meyectl[451]: self.set_config(camera_id)
Nov 11 07:26:37 raspberrypi meyectl[451]: File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 166, in wrapper
Nov 11 07:26:37 raspberrypi meyectl[451]: return func(self, *args, **kwargs)
Nov 11 07:26:37 raspberrypi meyectl[451]: File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 483, in set_config
Nov 11 07:26:37 raspberrypi meyectl[451]: set_camera_config(camera_id, ui_config, on_finish)
Nov 11 07:26:37 raspberrypi meyectl[451]: File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 326, in set_camera_config
Nov 11 07:26:37 raspberrypi meyectl[451]: local_config = config.motion_camera_ui_to_dict(ui_config, local_config)
Nov 11 07:26:37 raspberrypi meyectl[451]: File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 881, in motion_camera_ui_to_dict
Nov 11 07:26:37 raspberrypi meyectl[451]: if ui['despeckle_filter']:
Nov 11 07:26:37 raspberrypi meyectl[451]: KeyError: 'despeckle_filter'
Nov 11 07:26:37 raspberrypi meyectl[451]: ERROR: 500 POST /config/1/set/?_username=admin&_admin=true&_signature=699fa980545f3ab2bfd07d8213aa0519bb3a1cc9 (192.168.1.8) 85.84ms

ccrisan, 2016-11-11:
The latest motionEye is not compatible with the latest motionEyeOS. I have yet to release a new version of motionEyeOS integrating all the new changes. Sorry about that.

moonguar, 2016-11-11:
No problem.  I just wanted to make sure it wasnt something I did.  Any plans to update the OS version?  estimated date?

closed 2017-06-01 by bortek