motioneye-project/motioneyeos #593: Fast network camera snapshot url

jmai444, 2016-11-13:
A snapshot url does not appear anywhere in the configuration panel when the fast network camera option is enabled.

I'd like to use an http request to trigger a snapshot.  Possible?

ccrisan, 2016-11-13:
The current FNC implementation (based on streamEye) does not allow serving a single snapshot. I have opened an [issue](https://github.com/ccrisan/streameye/issues/8) on this matter but this is not a priority so don't expect it to be done any time soon.

closed 2017-06-01 by bortek

davimas, 2018-11-13:
Hey, any solution after 2 years?
I need a jpg snapshot url, an not will use a another System. Motioneyeos is great, and a read only System in the sdhc card > this is very importend for the life of the card......

bsvedin, 2023-02-24:
This still remains an issue
I would like to see this fixed

starbasessd, 2023-02-24:
I doubt it will be. Calin gave control over to new devs a year ago, and very little has gone forward. This ticket was opened 7 years ago and closed 5 years ago. There hasn't been any activity on the streameye in several years, either.