motioneye-project/motioneyeos #594: Network Share Causes No Camera Status after 5-10 minutes

johnelle, 2016-11-13:
I am trying to use the built-in network share support to a Synology NAS on the local net.  After a few minutes I still have a functional UI but there is s "no camera" icon on the display and the camera appears to be dead.   This happens on both wireless and wired connection (last test was wired only).  Running latest version on a Pi2 B+.

1)  Hitting the "test share" button fails
1)  If I reboot the Pi it starts working again (for a little while)
2)  If I change a character of the path and test/reconnect (i.e. apply) it reconnects fine
3)  I set the resolution to the lowest level and that has no effect.

I saw this with the previous motionpi software when I did my own mounting but it usually worked for 24 hours at a time.

Its as if one error causes the thing to lock and it won't work until you manually reconnect the share.

ccrisan, 2016-11-13:
Well if the Test Share button indicates an error, then you have an error with your share configuration. Please attach your motioneye.log file here so I can try to identify the issue.

johnelle, 2016-11-13:
The Test Share is fine when the NAS is first connected.  Files get written
to the NAS.  It works.

After about an hour the UI is displaying a "No Camera" icon and although
the UI still works fine, no images are coming from the camera (not sure how
why the camera has anything to do with the share connect but that is the
behavior I see).  If I force the re-connection to the NAS by rebooting or
changing the path (just the first letter cap/nocap) and then reconnecting
things start working again.  Functioning of the camera seems to be tied to
NAS connection failing.  Just reconnecting starts everything back up.

Essentially the NAS locks up after an hour or so and I have to reconnect to
get the camera working again.  However the minute I reconnect it is fine.

This does not happen if I write to local storage with all the same
settings.   Image size does not affect it.  Wireless vs. Wired does not
effect it.

Bottom Line:  the NAS connection is not reliable enough to use.

Pi2 B+ with V2 Camera.

johnelle, 2016-11-14:
...and simply switching back to local storage also causes it to start
working again.  The problem is that there are limited push options (no
email attachments) so for unattended use you would have to write a daemon
to pull things over the samba connection.

ccrisan, 2016-11-14:
Whenever there are some storage-related issues, the motion daemon (used by motionEye as a backend) will stop working correctly. This includes the streaming and therefore you get the No Camera indication in the UI.

For some reason, the network share fails after some time. It's very hard for me to investigate such a problem, especially since you mentioned it's sporadic.

I myself use a NAS with motionEyeOS but I never had stability problems with it. I could take a look over your log files if you want to attach them here, but I'm not promising anything.

johnelle, 2016-11-14:
Ok.  As I mentioned I had seen this with a different NAS and a different
network using the previous (motionpi) version.

I am happy to capture a log with it hooked to NAS since I am testing mode
anyway.  What is the location of the log file that you want me to forward?

ccrisan, 2016-11-14:
Well just download them all from the UI and attach them here. There's no sensible information in them so there's no need to worry about posting them here.

johnelle, 2016-11-14:
System rebooted at *2016-11-14 10:45:43  *

Locked up about *2016-11-14 10:56:26*

Log attached

*Network Settings*

​

johnelle, 2016-11-15:
I switched the share to the equivalent of anonymous access ("guest" on the
synology) so it can be accessed with no credentials.  Now I still get an
access error every 10 minutes or so but the camera stays working.

It appears to me that shares that require a login seem to have trouble
reconnecting from motioneye but shares that are open are able to reconnect.

Log attached (using the release from yesterday)

ccrisan, 2016-11-15:
@johnelle you have never really attached any log files. You're probably hitting reply by email, which will never include your attachments. Please *always* use the github issues page.

johnelle, 2016-11-16:
Here is my motioneye test board running with a local synology NAS using a share that requires authentication.

 System rebooted at 2016-11-14 10:45:43  

Locked up about 2016-11-14 10:56:26

Log attached

Network Settings
[image]

[motioneye (2) (1).txt](https://github.com/ccrisan/motioneyeos/files/594597/motioneye.2.1.txt)

johnelle, 2016-11-16:
I switched the share to the equivalent of anonymous access ("guest" on the synology) so it can be accessed with no credentials.  Now I still get an access error every 10 minutes or so but the camera stays working.  

It appears to me that shares that require a login seem to have trouble reconnecting from motioneye but shares that are open are able to reconnect.  

Log attached (using the release from yesterday).  Has been running 24 hours.

I would prefer to have the guest account set to inactive as I am one of those people that got attacked in the massive synology hostageware attack and am still smarting from that.

[motioneye (3) (1).txt](https://github.com/ccrisan/motioneyeos/files/594610/motioneye.3.1.txt)

ccrisan, 2016-11-16:
Please never use backslashes in share names. In Linux, all paths use forward slashes. Also, the share name must be a single word (i.e. path element). It should never contain slashes (either forward or backward).

See [this](https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#network-shares) for more details on how to configure your share.

johnelle, 2016-11-16:
Ok but if I follow your directions it doesn't work.

This syntax connects 

[image]

This syntax fails

[image]

ccrisan, 2016-11-16:
Beats me. What does the full path look like when you're browsing it from Windows (Windows Explorer).

johnelle, 2016-11-16:
Pasted directly from Windooz 10 Explorer

\\192.168.1.38\Media\bustercam

(the same on Ubuntu)

ccrisan, 2016-11-16:
Then the second screenshot (with the syntax that fails) is the correct one. I have no idea why it fails.

johnelle, 2016-11-16:
Well here's what I know:

1)  The "bad" syntax is the only thing that works in my case
2)  After I connect using the "bad" syntax it eventually fails.  The behavior is that it seems to disconnect every 10 minutes or so regardless.  If there are any credentials involved it cannot reconnect and locks up.

The only way I can get it to work reliably is to remove the credentials on the synology side so it can do its disconnect/reconnect thing every few minutes.  

The NAS is a Synology DS214 running the latest version of DSM.  In case you don't know the Synology is essentially a Linux box inside.

donnlee, 2016-12-15:
I did not have any luck with mounting SMB (cifs) shares owned by a Win7 pc.  Authentication was horrible to troubleshoot; had to give up.  Unauth (no pw) worked for a few mins or a few hours, then file transfers would stop; had to restart the Win7 service to get it to work again (for another few mins or hours).
Finally, gave up on SMB (it's always been a broken protocol in my experience) and switched to NFS mounts.  Has been solid ever since.

My NFS server syncs with Dropbox, so I have cloud storage: in case burglar disables the cam and server, I still can confirm a break-in and call local police.

dels78, 2016-12-27:
@donnlee, how did you mount your NFS drive? I tried to do that from fstab directly, then from the UI, but it won't work for me... 

Thanks for the help

donnlee, 2017-01-11:
@dels78, I do the nfs mount like this:
```
mount -t nfs -o nolock,vers=2 192.168.1.113:/home/donn/srv/dropbox/motioneye/nfs_cam1 /data/nfs
```

To make the mount happen at bootup, I just copied a simple init.d script (stole from S62smb) and made it execute the 'mount' command.
Note, motioneyeos doesn't have runlevels so all startup scripts are in /etc/init.d/
Named the script S65mount_nfs:

```
#!/bin/bash

test -n "$os_version" || source /etc/init.d/base
test -n "$os_debug" || source /etc/init.d/conf

has_net_conn || exit 0

start() {
    msg_begin "Mounting donn nfs share."
    mount -t nfs -o nolock,vers=2 192.168.1.113:/home/donn/srv/dropbox/motioneye/nfs_cam1 /data/nfs
    test $? == 0 && msg_done || msg_fail
    # Disable TCP Timestamps option, else can't connect to my VPS server for SMTP.
    echo 0 > /proc/sys/net/ipv4/tcp_timestamps
}

stop() {
    msg_begin "Nothing really to stop here."
    echo "..."
    test $? == 0 && msg_done || msg_fail
}

case "$1" in
    start)
        start
        ;;

    stop)
        stop
        ;;

    restart)
        stop
        start
        ;;

    *)
        echo "Usage: $0 {start|stop|restart}"
        exit 1
esac

exit $?
```

But the root filesystem is mounted read-only. So to write this to /etc/init.d/
I first create the file in /data/ then copy to /etc/init.d/ with a little script like so:

```
[root@meye-936cab38 data]# cat run.sh
#!/bin/bash

/bin/mkdir -pv /data/nfs
/bin/chmod 775 /data/nfs
/bin/sync
/bin/mount -o remount,rw /
/bin/sync
/bin/cp /data/S64mount_nfs /etc/init.d/
/bin/sync
/bin/mount -o remount,ro /
/bin/sync
```

I had a corrupted root filesystem in the past that killed my motioneyeOS, so that's why I'm hyper-paranoid with all the sync's.

rpiuser, 2017-02-14:
I'm having the exact same behaviour on a Raspberry PI Zero, using motioneyeos-raspberrypi-20161125.img, I also have it running on a PI2 without any issues using motioneyeos-raspberrypi2-20161212.img - was this resolved ?

benotje, 2017-02-16:
Network Share Causes No Camera Status after 5-10 minutes
i have een raspberry 2 with motioneyeos-raspberrypi-20161125.img
i have synology nas for the data 
works nice for 5 a 10 min than no connection

when I send the data to the SD motioneye He continues to work 

What can I do to fix this

closed 2017-06-01 by bortek