motioneye-project/motioneyeos #595: Notification email failed: character mapping must return integer, None or unicode!

kuckuck8090, 2016-11-13:
Hi all,

I have change my system from a banana pi to a raspberry pi 2.
I have installed on raspberry the current version 20161027 and do the setup.
The raspberry gets the signal from a network camera via mjpeg stream and it works.
Only if I make the setup for email notification, I will get an Error message (see topic).
I don't know what is wrong. It was blank installation.
For your information: we previous installation works fine with my hoster. I have only change the password(all character) and try also with AlphaNumerical only.

Thanks in advanced.

ccrisan, 2016-11-13:
Please attach your log files here.

ccrisan, 2016-11-13:
The sendmail module complains indeed when performing the authentication. Can you tell me all of your email configuration details, except for the password? A screenshot of how the email settings section looks like would be great.

ccrisan, 2016-11-13:
It seems as though you're affected by [this](http://bugs.python.org/issue5285) bug. Unicode passwords are not accepted, it seems. Please try using a plain ascii password.

kuckuck8090, 2016-11-13:
Hi,
I have change the password that is conform to ascii, without sucess.
[image]

ccrisan, 2016-11-13:
Then I'm sorry but I have no idea how to help you with this one.

kuckuck8090, 2016-11-13:
Thanks for your help.
I have try also to set the password with "1" and I get the same message.
It's look like a bug, because he doesn't accept any passwords without Error message.
I look forward for the next version :) and in the meantime I will try a previous version.

donnlee, 2016-11-25:
I'm blocked by this error too.
My smtp password is simple alpha and number chars.

ccrisan, 2016-11-25:
@donnlee do those characters include non-ascii (accented) letters?

donnlee, 2016-11-25:
@ccrisan No, no accented letters at all.  I live in the US and my smtp password consists of standard US keyboard letters a-z and 0-9.
I should add that I have no problems configuring this to work with Gmail, which is also TLS.
But when I try my smtp on my webhosting provider, I get this "character mapping" error.
The howto instructions from my webhosting provider is the same as Gmail, just different server name and login.

donnlee, 2016-12-02:
I built a new Postfix SMTP server recently. When I point motioneyeOS to use it for email notifications (TLS enabled), I do not get the "character mapping" error.
So maybe it is an incompatibility with just some SMTP servers/code.

I did hit a new problem though. MotioneyeOS was not able to pass the basic TCP 3-way handshake with my server (SMTP, ssh, http, or any other tcp connection).
I had to disable TCP Timestamps option on MotioneyeOS to workaround that problem.
Everything is working great now.

ccrisan, 2016-12-02:
@donnlee can you please elaborate a bit on this TCP timestamps issue? This is the first time I hear about it and if there's anything I can do to fix this in the next release, I would gladly do.

donnlee, 2016-12-08:
@ccrisan Thanks for your efforts to make this project compatible in as many situations as possible.
I observed the following:
I used my Macbook and the Apple "Mail" app to test my postfix mail server (ubuntu 14.04). Worked fine. The Mail app connects to the postfix server on port 587, submits its mail, and disconnects. Postfix delivered the mail to Gmail.
When I had motioneyeOS send the test email, the motioneye web-ui displayed an error msg that the connection timed out (email failed).

Then I setup a webserver on my postfix server and created a simple webhook.
Again, my macbook has no problems hitting the webhook.
MotioneyeOS couldn't connect / load the URL. That was weird.  curl/HTTP on port 80 should definitely work.

So I ran tcpdump on the smtp port 587 connection.
I saw that motioneyeOS's SYN for the tcp connection was ignored by the postfix server.
Actual tcpdump (IP addresses anonymized) on the postfix server:

17:35:53.006049 IP 1.1.1.1.1612 > 2.2.2.2.587: Flags [S], seq 4252930093, win 29200, options [mss 1460,sackOK,TS val 8692689
 ecr 0,nop,wscale 7], length 0
17:35:54.941537 IP 1.1.1.1.1612 > 2.2.2.2.587: Flags [S], seq 4252930093, win 29200, options [mss 1460,sackOK,TS val 8692889
 ecr 0,nop,wscale 7], length 0
17:35:58.953487 IP 1.1.1.1.1612 > 2.2.2.2.587: Flags [S], seq 4252930093, win 29200, options [mss 1460,sackOK,TS val 8693290
 ecr 0,nop,wscale 7], length 0

HTTP conns from motioneyeOS to the webserver was similar. SYNs arrive at webserver but webserver doesn't reply with SYN-ACK.  Weird.

Note, my macbook has tcp timestamps enabled and has no problems. Here's a tcpdump capture ("TS val"):
17:57:02.108730 IP macbook_ip.1772 > postfix_server_ip.587: Flags [S], seq 3874264686, win 65535, options [mss 1460,nop,wscale 5,nop,nop,TS val 663026692 ecr 0,sackOK,eol], length 0

Then I found:
http://serverfault.com/questions/235965/why-would-a-server-not-send-a-syn-ack-packet-in-response-to-a-syn-packet

I disabled tcp timestamp option on motioneyeOS:

```
[root@meye-936cab38 ~]# cat /proc/sys/net/ipv4/tcp_timestamps
1
[root@meye-936cab38 ~]# echo 0 > /proc/sys/net/ipv4/tcp_timestamps
[root@meye-936cab38 ~]# cat /proc/sys/net/ipv4/tcp_timestamps
0
```

And all my problems went away.
To make the workaround permanent, I added "echo 0 ..." to an init script.

I didn't investigate further as I wanted to tune other parts of my motioneye cam :)

Voodoofly, 2016-12-10:
I have the same error when i configure Motioneye an when i push the test Button "character mapping must return integer, None or unicode"   

But i ignore it, unplugged short the power, and test again

The error at test button comes allways, but sending pictures to email works fine

I think theres only a bug in the Test Mode

bortek, 2017-06-01:
Closing this issue due to inactivity. If stil interested please post your question in the forum page.

closed 2017-06-01 by bortek