motioneye-project/motioneyeos #596: URL DIRECTLY TO FULL SCREEN VIDEO VIEW

muaythai2006, 2016-11-13:
Hi,
Is there a away to enter the url on chrome browser to view the video in full screen.
example. http://user:test@192.168.1.100 . right now inorder to view the video in full screen. O have to login then click the full screen in video video.
thanks

Mattncb, 2016-12-08:
I would also be interested in this, did you make any progress? 
I know it has a security risk, but Im looking to make a URL string that includes USR and PWD in order to log straight in the stream. You could do this in RPICam but doesnt look like you can from all the combos ive tried so far.

ccrisan, 2016-12-10:
There already are a few possibilities to access a stream in full-screen, depending on what you mean by full screen:

1. You have the streaming URL (copy it from the Settings Panel/Video Streaming) that can be viewed directly in a browser or a player that supports MJPEG. It is protected by the normal user password, so if you set it to an empty string, it won't require a password.

2. You have the embed URL that is protected by the normal user password as well.

3. You have the snapshot URL that includes the authentication information in the URL itself so it won't ask for credentials. However it won't refresh by itself.

4. The next release of motionEye(OS) will allow saving and remembering credentials without the need to reenter them each time.

What you can't do is put the browser in full screen by simply accessing a URL.

closed 2016-12-10 by ccrisan