motioneye-project/motioneyeos #597: Upgrade to 20161114

duckietm, 2016-11-15:
I have upgraded to the 20161114  and my Raspb. camera is not beeing detected anymore !
How do i go back to the older version without opening the camra again as it is outside and it takes me 2 hours to open it :(

Nov 15 06:47:54 meye-ab11a962 user.warn motion-mrdave: [1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD

[Thu Jan  1 00:00:01 1970] bcm2835-v4l2: scene mode selected 0, was 0
[Thu Jan  1 00:00:01 1970] bcm2835-v4l2: V4L2 device registered as video0 - stills mode > 2592x1944
[Thu Jan  1 00:00:01 1970] bcm2835-v4l2: Broadcom 2835 MMAL video capture ver 0.0.2 loaded.
[Thu Jan  1 00:00:01 1970] bcm2835-wdt 3f100000.watchdog: Broadcom BCM2835 watchdog timer

ccrisan, 2016-11-15:
You can't. Please attach your log so we can identify the problem.

duckietm, 2016-11-15:
[logs.zip](https://github.com/ccrisan/motioneyeos/files/591295/logs.zip)

duckietm, 2016-11-15:
attached

duckietm, 2016-11-15:
if i do 
cat /dev/video0
i do see that the device is responding and output is put at the putty session (of course  unreadble )

duckietm, 2016-11-15:
I opened up the camera and placed back version motioneyeos-raspberrypi2-20161027
Star it up and works straight away
Then did the upgrade again, same problem.... so i am back @20161027

duckietm, 2016-11-15:
I think i found the problem, in the Motion Binary i got : 
/usr/bin/motion-mrdave when i change this to /usr/bin/motion it works
In the old version this works fine, but in the latest build not.

ccrisan, 2016-11-15:
@duckietm sorry for the late answer and thanks for keeping us posted. Indeed, there's no motion-mrdave anymore, as it evolved into the new motion 4.0. I'll put a warning in the release notes so that users having explicitly set their motion binary will know to change it back (you had it changed manually to `motion-mrdave`, right?)

duckietm, 2016-11-16:
No i had to set it to : /usr/bin/motion

ccrisan, 2016-11-16:
That's strange. How did it get changed to `/usr/bin/motion-mrdave` then? How did you actually find the problem?

duckietm, 2016-11-16:
i re-imaged and did a restore from backup then i think the setting was there.
I use it for 2 years now so somewhere this has been added :)

ccrisan, 2016-11-16:
Alright, thanks! I'll close the issue then.

closed 2016-11-16 by ccrisan