motioneye-project/motioneyeos #600: Capturing multiple still frames/sec?

c0lpanic, 2016-11-16:
I'm trying to capture as many frames as possible. As such, I have turned off collecting video files and am just collecting still images. However, there can only be one image with each timestamped name. Is there a way to change the naming structure so I can capture multiple stills in one second? Also, is this the best way to get the highest framerate? Thank you!

ccrisan, 2016-11-16:
Yes, add a %q in the image name pattern.

Chiny91, 2016-12-04:
Ha yes, which reminds me just how long (far too long) it took me to find [this wiki page](http://www.lavrsen.dk/foswiki/bin/view/Motion/AdvancedFilenames) a while back.   Thanks for the %q reminder.

Re-reading the wiki page; I do notice that page has a security warning suggesting _that the motion control port is not available from the outside_.   That doesn't appear to be the case at present with motionEyeOS.   With well-reported DDOS attacks being launched from low power devices, IPcams etc, it might be approaching the time to have a view re motionEyeOS security.

scoobyd00, 2016-12-09:
Maybe worth adding this %q to the wiki or even the program?.

I didnt know this existed and i was saving only still images. It was working well enough and capturing people as they walked down my driveway but i was getting about 4 frames, with the %q option im getting 20+ frames and capture every movement.

ccrisan, 2016-12-09:
I added a FAQ entry: https://github.com/ccrisan/motioneyeos/wiki/FAQ#how-can-i-make-motioneyeos-save-more-than-one-picture-in-a-second

ccrisan, 2016-12-09:
https://github.com/ccrisan/motioneye/commit/0a315ef637be5ca96c700c7bb5af690954608acd adds more special tokens to the descriptions of the various options.

closed 2016-12-09 by ccrisan

ccrisan, 2016-12-09:
@Chiny91 

> Re-reading the wiki page; I do notice that page has a security warning suggesting that the motion control port is not available from the outside. That doesn't appear to be the case at present with motionEyeOS. With well-reported DDOS attacks being launched from low power devices, IPcams etc, it might be approaching the time to publish a view re motionEyeOS security.

Control port is the port used to control the motion daemon itself and it is set by motionEye to 7999. Do you want to say that motion listens on 7999 on all interfaces?

svoket, 2016-12-09:
What is the path to the file please? 

-----

Chiny91, 2016-12-13:
Perhaps not well phrased; I moved from the motion control port to motionEyeOS in general without warning.

I did _nmap_ motionEyeOS a while ago and found it responded on a number of ports, about 8, ISTR.  It did make me quickly look at a security on motionEyeOS and noted some good stuff (example, updated ntpd without monlist bug) and less obviously so (example, shipped with admin but no password).

Currently in the UK (and maybe elsewhere), there is a lot of concern re DDoS attacks from the _Internet of Things_. This has impacted the RPi (Raspbian) causing [changes](https://www.raspberrypi.org/blog/a-security-update-for-raspbian-pixel/) with that article giving a flavour of concerns.

Obviously over-enthusiastic security would make motionEyeOS hard to use and defeat its purpose so, as ever, a balancing act.

ccrisan, 2016-12-13:
@Chiny91 I'd be happy to (let others help me) improve the security of motionEyeOS. I'm open to suggestions and to pull requests :)

Chiny91, 2016-12-13:
Ah, oops.   Must learn to keep quiet :neutral_face: