motioneye-project/motioneyeos #601: Pixelated movies

Oreron, 2016-11-17:
Hi

I have a strange problem running MotionEyeOS on a Raspberry Pi 2. Whatever resolution/framerate/quality I set, the movies become very pixelated. See attached image. Still images looks fine.

I have another camera running off an old Raspberry Pi 1 (manually configured with "motion"). The movies produced with this are all OK (not pixelated).

Both setups use the official Raspberry Pi cameras.

Both produce .ogg files (I need to play them from a web page). I have noticed that the MotionEyeOS produce very small .ogg files (under 20k) files compared to the manually configured camera (about 100k).

Also, I have tried to add custom parameters to "Extra motion options", but some of them will not save and are gone after "Apply"...

Any ideas?

[image]

ccrisan, 2016-11-17:
What's your movie quality setting?

Oreron, 2016-11-17:
Right now 100% with framerate 5.

ccrisan, 2016-11-17:
What version?

Oreron, 2016-11-17:
20161027 - The latest?

ccrisan, 2016-11-17:
Could you please try the prerelease 20161114?

Oreron, 2016-11-17:
Absolutely. Is it possible to install it from motionEyeOS or do I need to download and create a new SD card?

Oreron, 2016-11-17:
Never mind. Installing now...

Oreron, 2016-11-17:
The install worked well. Two new issues after installation:

1. The output file is .avi, even if I selected .ogg
2. The movie starts all black, then there is the same frame playing the entire movie

ccrisan, 2016-11-17:
Please attach your config files here, after *removing any sensitive data*. I'm not sure about ogg videos but I remember seeing a post saying that it's been deprecated and no longer supported by motion. Not sure, though.

Oreron, 2016-11-17:
OK, then we can disregard .ogg. I changed to .mp4 and it seems to work fine for now. Pixelation gone :-) and it plays fairly smooth from my webpage.

Thanks for your help, and thanks for creating this fantastic OS :-)

closed 2017-06-01 by bortek