motioneye-project/motioneyeos #605: PIR Sensors ModMyPi Box

nextgen1979, 2016-11-18:
Hello, 

I possible to use the modmypi box with the PIR Sensors included with motioneye os ? 

Thanks in advance 

Nicolas

db414813, 2016-11-30:
This is a cool feature and I believe it's planned. Sooner or later.

ril3y, 2017-02-18:
@ccrisan I would be willing to implement this but I am not really sure where to start.  Can you give me a suggestion on where you would start poking around if you were going to do this?  Turns out these PIR modules just output a HIGH ttl level for 2 seconds then go a back down to low.  

Pretty simple.  It might be a good enough of an option to put "take picture on motion" or "run this script" when this GPIO goes high.  Thoughts?

Willing to help just not familiar enough with the project.

fti7, 2017-03-20:
Is there any command we could execute to trigger MotionEyeOSs internal "Motion Detection" Mechanism (Recording Pictures/Videos/whatever configured)?

An Daemon which polls an GPIO Pin + Executes this Command would be very easy

ccrisan, 2017-03-21:
Not yet.

cephasiii, 2017-03-27:
i am interested in this feature as well. looking forward to seeing it implemented.

hellyberry, 2017-04-14:
This will be very usful with a generic interface to use other sensors also.

jasaw, 2017-06-01:
See issue #842 .

closed 2017-06-01 by bortek