motioneye-project/motioneyeos #608: NexGadget Mini Cam - Authentication failed

gioluca2, 2016-11-19:
I use motioneyeos-20161114 on my raspberry pi 2 and I have this issue with NexGadget Mini Cam (https://www.amazon.it/gp/product/B01LPJZVZS/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1):
when I insert the link (192.168.1.101), username and password I have "authentication failed" error message:
[image]

When I open http://192.168.1.101/ into browser I have this screen:
[image]

I also try to add the two direct links (http://192.168.1.101/main.html?id=1&playtype=iv and http://192.168.1.101/main.html?id=1&playtype=fv) but without success.
It doesn't work http://192.168.1.101:554 (RTSP port with and without authentication request) either.

Please help me

bortek, 2017-06-01:
Closing this issue due to interest. If still interested please post your question in the forum page.

closed 2017-06-01 by bortek

thesatishk, 2017-06-21:
@gioluca2 did you manage to get this camera work with the pi?

thesatishk, 2017-06-25:
https://www.ispyconnect.com/man.aspx?n=iSmartViewPro#

Here's the info about the rtsp stream for this camera.