motioneye-project/motioneyeos #610: Unable to connect to pi with static IP address

tdodson90, 2016-11-21:
This seems to happen frequently. I can't get to the WebUI, or SSH, for some reason, the port is being refused. The strange thing is, the pi is still uploading photos to dropbox. Once I reboot the pi, all comes back up.

What log should I be looking at?

ccrisan, 2016-11-21:
Your log files could be telling a lot about your problem.

tdodson90, 2016-11-21:
Which log specifically should I look at? I can post the logs here if you think you could take a peek.

ccrisan, 2016-11-22:
Please post all of them here.

tdodson90, 2016-11-22:
Attached logs

[motion logs.zip](https://github.com/ccrisan/motioneyeos/files/606552/motion.logs.zip)

ccrisan, 2016-11-22:
The only problem I can see in your logs is that of an internet connectivity every now and then. Because of that, the system date/time is wrong (unset). Without a correct system date/time many things can go wrong.

tdodson90, 2016-11-22:
Ah ok. I have set the system to use an ubuntu NTP server. Hopefully that corrects the issue. Just out of curiosity which log did you see the network connection issues? Could you send me an example, that way I can check in the future? :)

ccrisan, 2016-11-22:
boot.log:

    Setting current date using http: failed

dmesg.log:

    [Wed Dec 31 18:00:28 1969] system date is 1969-12-31 18:00:29

motioneye.log:

    1969-12-31 18:15:06: [motioneye]    ERROR: failed to upload file "/data/output/Camera1/1969-12-31/18-15-00.jpg" with service dropbox: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:590)>

closed 2016-11-22 by ccrisan